osx-cross/homebrew-avr #131: Build Failed

Mattimand, 2019-10-19:
I used to run with larsimmisch's XAVR for Xcode until I got a new Mac and had to install OSX-Cross/AVR instead. The old version used to work fine, but the new version won't seem to build my projects, let alone upload them. I tried downloading the official Arduino IDE and run a simple blink program, which seemed to both build and upload fine, to the right board, through the right port and everything. Using the same settings in Xcode, I haven't had the same luck.

Here are the last lines of my error report:

```
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00

avrdude done.  Thank you.

make: *** [program] Error 1
Command ExternalBuildToolExecution failed with a nonzero exit code
```

ladislas, 2019-10-20:
Could you share your system info with us please? OS, Xcode version, brew version, avr-gcc/avrdude version, etc.

XAVR and osx-cross/avr are not at all related:

- https://github.com/jawher/xavr is an xcode template/projet to write avr code in Xcode -- it has not been updated since 2016
- https://github.com/osx-cross/homebrew-avr (this repo) is just a way to install avr-gcc and tools to build avr code, it doesn't do anything else

From your error code, the issue seems to come from avrdude. Maybe you haven't configure the right board/port/speed or the board/port is busy with something else (Serial console from Arduino maybe?)

Mattimand, 2019-10-24:
Thank you for the fast reply (something I can't say to have reciprocated)!

OS: **macOS 10.13.6 High Sierra**
Xcode: Version 10.1 (10B61)

And I pretty much just installed brew and all the rest last week, so they should be up to date. Though, a command to check the version would be neat to know in the future.

Running a `brew doctor` I get the following:
`Your system is ready to brew.`

I figured XAVR and osx-cross/avr weren't different when I discovered the latter. The thing is that I installed XAVR on my MacBook last semester (January, 2019) and it worked smoothly with no issues. Then this semester I tried helping a colleague install it on his MacBook, however he got a ton of errors, and so when I got my new iMac and wanted to install XAVR, I realised that the repository was no longer tappable. After digging some research, I figured that XAVR had been abandoned and that osx-cross/avr was the way to go.

So I wiped my system and completely erased my hard drive and installed Homebrew on a fresh OS X 10.13.6 High Sierra with Xcode 10.1 (10B61) and followed the three commands to install osx-cross/avr. The template worked. The colours worked. I could ⌘-click all the definitions in my code.

But for some reason, my builds would constantly fail. And I tried using different values for `AVRDUDE_PROGRAMMER` such as `arduino` and `avrispmkII`.

`AVRDUDE_PROGRAMMER = arduino` yields
```
ExternalBuildToolExecution All (in target: All)
    cd /Users/Matt/Documents/DTU\ -\ Bachelor\'s\ Degree/62634\ -\ Data\ Technique\ and\ Programming/Assignment\ 1
    export ACTION=
    export ALTERNATE_GROUP=staff
    export ALTERNATE_MODE=u+w,go-w,a+rX
    export ALTERNATE_OWNER=Matt
    export ALWAYS_EMBED_SWIFT_STANDARD_LIBRARIES=NO
    export ALWAYS_SEARCH_USER_PATHS=NO
    export APPLE_INTERNAL_DEVELOPER_DIR=/AppleInternal/Developer
    export APPLE_INTERNAL_DIR=/AppleInternal
    export APPLE_INTERNAL_DOCUMENTATION_DIR=/AppleInternal/Documentation
    export APPLE_INTERNAL_LIBRARY_DIR=/AppleInternal/Library
    export APPLE_INTERNAL_TOOLS=/AppleInternal/Developer/Tools
    export APPLICATION_EXTENSION_API_ONLY=NO
    export APPLY_RULES_IN_COPY_FILES=NO
    export ARCHS=x86_64
    export ARCHS_STANDARD=x86_64
    export ARCHS_STANDARD_32_64_BIT="x86_64 i386"
    export ARCHS_STANDARD_32_BIT=i386
    export ARCHS_STANDARD_64_BIT=x86_64
    export ARCHS_STANDARD_INCLUDING_64_BIT=x86_64
    export AVAILABLE_PLATFORMS="appletvos appletvsimulator iphoneos iphonesimulator macosx watchos watchsimulator"
    export BITCODE_GENERATION_MODE=marker
    export BUILD_ACTIVE_RESOURCES_ONLY=NO
    export BUILD_COMPONENTS="headers build"
    export BUILD_DIR=/Users/Matt/Library/Developer/Xcode/DerivedData/62734_-_Data_Technique_and_Programming-coiteabhkufhvmdyfvyksvzlpcxi/Build/Products
    export BUILD_ROOT=/Users/Matt/Library/Developer/Xcode/DerivedData/62734_-_Data_Technique_and_Programming-coiteabhkufhvmdyfvyksvzlpcxi/Build/Products
    export BUILD_STYLE=
    export BUILD_VARIANTS=normal
    export BUILT_PRODUCTS_DIR=/Users/Matt/Library/Developer/Xcode/DerivedData/62734_-_Data_Technique_and_Programming-coiteabhkufhvmdyfvyksvzlpcxi/Build/Products/Debug
    export CACHE_ROOT=/var/folders/9r/dzx3jkns407ds435pyhjr8cr0000gn/C/com.apple.DeveloperTools/10.1-10B61/Xcode
    export CLANG_ANALYZER_NONNULL=YES
    export CLANG_ANALYZER_NUMBER_OBJECT_CONVERSION=YES_AGGRESSIVE
    export CLANG_CXX_LANGUAGE_STANDARD=gnu++14
    export CLANG_CXX_LIBRARY=libc++
    export CLANG_ENABLE_MODULES=YES
    export CLANG_ENABLE_OBJC_ARC=YES
    export CLANG_ENABLE_OBJC_WEAK=YES
    export CLANG_MODULES_BUILD_SESSION_FILE=/Users/Matt/Library/Developer/Xcode/DerivedData/ModuleCache.noindex/Session.modulevalidation
    export CLANG_WARN_BLOCK_CAPTURE_AUTORELEASING=YES
    export CLANG_WARN_BOOL_CONVERSION=YES
    export CLANG_WARN_COMMA=YES
    export CLANG_WARN_CONSTANT_CONVERSION=YES
    export CLANG_WARN_DEPRECATED_OBJC_IMPLEMENTATIONS=YES
    export CLANG_WARN_DIRECT_OBJC_ISA_USAGE=YES_ERROR
    export CLANG_WARN_DOCUMENTATION_COMMENTS=YES
    export CLANG_WARN_EMPTY_BODY=YES
    export CLANG_WARN_ENUM_CONVERSION=YES
    export CLANG_WARN_INFINITE_RECURSION=YES
    export CLANG_WARN_INT_CONVERSION=YES
    export CLANG_WARN_NON_LITERAL_NULL_CONVERSION=YES
    export CLANG_WARN_OBJC_IMPLICIT_RETAIN_SELF=YES
[... 316 lines cut ...]
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00

avrdude done.  Thank you.

make: *** [program] Error 1
Command ExternalBuildToolExecution failed with a nonzero exit code
```

Changing to `AVRDUDE_PROGRAMMER = avrispmkII` technically _succeeded the build_, however didn't upload anything to the board.

Switching to the Arduino IDE (with no colours for DDRs and PORTs) and building it seemed to work, so long as the correct board under Tools, and it uploaded perfectly -- the very same code I just copied over from Xcode. So my computer is clearly _able_ to build, compile, and upload. It's just Xcode that won't upload.

Let me know if you need any more details.

And sorry for the not-so-fast-as-your reply!

Mattimand, 2019-10-24:
It's worth noting that, by accidentally selecting the wrong board in the Arduino IDE I got the same error as in Xcode. So perhaps Xcode isn't recognising the `AVRDUDE_PROGRAMMER`?

Mattimand, 2019-10-24:
Food for thought: Could Xcode be lacking a target to upload to? I just see "iMac" under all profiles: All, Upload, Build, Clean.

ladislas, 2019-10-24:
I have never used Xcode to do what you are doing, neither have I used the xavr project template, so I can not help you and it is outside of the scope of this repository.

On a personal note, I'm not sure Xcode is best suited for what you're trying to do. You should take a look at [sudar's Arduino Makefile]() and my [Bare Arduino Project]() (it's using the aforementioned makefile) and let's you write C/C++ code for avr with or without using Arduino.

You can use VSCode or Vim or SublimeText to edit the code and you'll only need to set the project with the right flags do have autocompletion. Bare Arduino Project gives an example with Vim and YouCompeteMe.

That being said, in your case the issue seems to be with avrdude. Try running the following command from the command line. It's working for me.

```
avrdude -v -p atmega2560 -D -c wiring -b 115200 -P /dev/tty.usbmodem* -U flash:w:Builds/main.hex
```

Mattimand, 2019-10-24:
Thanks a lot!

1. Pasting your line in the Terminal worked smoothly.
2. Finding the same line in the MakeFile in my project and correcting it seemed to also solve the issue with Xcode's upload issue.

Pasting an excerpt from my MakeFile (if anyone should have the same issue):
```
#---------------- Programming Options (avrdude) ----------------

# Programming hardware: alf avr910 avrisp bascom bsd 
# dt006 pavr picoweb pony-stk200 sp12 stk200 stk500
#
# Type: avrdude -c ?
# to get a full listing.
#

AVRDUDE_WRITE_FLASH = -U flash:w:$(OBJDIR)/$(TARGET).hex
#AVRDUDE_WRITE_EEPROM = -U eeprom:w:$(TARGET).eep

# Uncomment the following if you want avrdude's erase cycle counter.
# Note that this counter needs to be initialized first using -Yn,
# see avrdude manual.
#AVRDUDE_ERASE_COUNTER = -y

# Uncomment the following if you do /not/ wish a verification to be
# performed after programming the device.
#AVRDUDE_NO_VERIFY = -V

# Increase verbosity level.  Please use this when submitting bug
# reports about avrdude. See <http://savannah.nongnu.org/projects/avrdude> 
# to submit bug reports.
AVRDUDE_VERBOSE = -v -v

AVRDUDE_FLAGS = -p $(MCU) -D -c $(AVRDUDE_PROGRAMMER) -b 115200 -P $(AVRDUDE_PORT) $(AVRDUDE_WRITE_FLASH)
AVRDUDE_FLAGS += $(AVRDUDE_NO_VERIFY)
AVRDUDE_FLAGS += $(AVRDUDE_VERBOSE)
AVRDUDE_FLAGS += $(AVRDUDE_ERASE_COUNTER)
```

So I simply had to swap this line:
`AVRDUDE_FLAGS = -p $(MCU) -P $(AVRDUDE_PORT) -c $(AVRDUDE_PROGRAMMER)`

For this line:
`AVRDUDE_FLAGS = -p $(MCU) -D -c $(AVRDUDE_PROGRAMMER) -b 115200 -P`

_Build Succeeded._

As to my friend's issue, however, I find it curious that when I run `brew tap osx-cross/avr` the xavr project template gets installed fine … but when he runs the same command, opening Xcode he finds no new template. I'm not sure whether I should start a separate thread for this, in case others might have the same issue?

Again, thank you for the line. Much appreciated.

ladislas, 2019-10-24:
I'm glad you solved the issue! :)

> As to my friend's issue, however, I find it curious that when I run brew tap osx-cross/avr the xavr project template gets installed fine [...]

There is no Xcode project template in this repository. You must have gotten it from somewhere else that is not related to osx-cross/avr.

Closing the issue now as it has been resolved.

closed 2019-10-24 by ladislas