osx-cross/homebrew-avr #132: Xcode Template??

Mattimand, 2019-10-25:
Hi again,

I wanted to hear if you could explain to me why an Xcode template is installed when I run:
```
brew tap osx-cross/avr
```

You said no template is associated with this repository. And I got a wholely new iMac and erased the entire hard drive before installing anything on it. Installing Xcode _without_ the template, and after this line very specifically `brew tap osx-cross/avr`, gave me a new template in Xcode called "xavr" that I cannot explain where came from if not this repository.

And the strangest thing is that, while this template came with `brew tap osx-cross/avr` on my MacBook as well, my colleagues MacBook installs it without a template.

I haven't downloaded _anything_. I just followed your very specific instructions, and the template appeared suddenly after the line `brew tap osx-cross/avr`.

ladislas, 2019-10-25:
> I wanted to hear if you could explain to me why an Xcode template is installed when I run

I have absolutely no idea.

Are you using Time Machine? Or any sort of backup method that could have reinstalled the template on your new mac?

Can you share some screenshots from xcode or anything that could be useful? The steps you took? Can you share the output of `brew list`, `brew cask list` and `brew doctor`?

> And the strangest thing is that, while this template came with brew tap osx-cross/avr on my MacBook as well, my colleague's MacBook installs it without a template, just like you said it would.

This is the less strange thing of all, this is exactly what is supposed to happen.

ladislas, 2019-11-03:
Closing the issue now as it's clearly not related to osx-cross/avr

closed 2019-11-03 by ladislas