osx-cross/homebrew-avr #136: avr-gcc is broken due to isl 0.22 update

drashna, 2019-12-31:
Brew updated isl to 0.22, and avr-gcc (8, and 9, at least) require 0.21. 

```
dyld: Library not loaded: /usr/local/opt/isl/lib/libisl.21.dylib
  Referenced from: /usr/local/Cellar/avr-gcc/9.2.0/libexec/gcc/avr/9.2.0/cc1
  Reason: image not found
avr-gcc: internal compiler error: Abort trap: 6 signal terminated program cc1
```

ubidefeo, 2019-12-31:
boom!
same here :(

ubidefeo, 2019-12-31:
temporary solution, @drashna
you might need some extra tools, I have a ton of development stuff on my machine.
* go to http://isl.gforge.inria.fr
* download the 0.21 (.xz)
* unpack it (double click)
* enter the directory in terminal
* `./configure`
* `make check`
if everything works you'll get something like this
```
PASS: isl_test
PASS: codegen_test.sh
PASS: pip_test.sh
PASS: bound_test.sh
PASS: isl_test_int
PASS: flow_test.sh
PASS: schedule_test.sh
PASS: isl_test_cpp
============================================================================
Testsuite summary for isl 0.21
============================================================================
# TOTAL: 8
# PASS:  8
# SKIP:  0
# XFAIL: 0
# FAIL:  0
# XPASS: 0
# ERROR: 0
============================================================================
Making check in doc
make[2]: Nothing to be done for `check'.
```

if tests are good, just go `make install`
if it all works out just `brew uninstall --ignore-dependencies isl`
after that you won't have that issue anymore
it fixed mine on Mojave, at least :)

drashna, 2019-12-31:
I just forcibly rolled back core, and disabled autoupdate.   Your solution is better, though.  However, neither are "good" solutions.

ladislas, 2019-12-31:
I'm out on holiday with my family with no computer. I'll update the formula when I get back next Monday.

drashna, 2019-12-31:
No worries, and enjoy your holiday!

darthburrito, 2020-01-01:
updating the reference worked for me
```
brew install isl
install_name_tool -change '/usr/local/opt/isl/lib/libisl.21.dylib' /usr/local/opt/isl/lib/libisl.dylib /usr/local/Cellar/avr-gcc/9.2.0/libexec/gcc/avr/9.2.0/cc1
```

ubidefeo, 2020-01-01:
@darthburrito 
I saw this as a possible fix on some other place, but it's an even dirtier trick in my opinion :D

drashna, 2020-01-02:
@darthburrito you may also need to do that for `lto1` as well.  At least, I have had to.

ladislas, 2020-01-05:
@drashna @ubidefeo I think the easiest solution for now is to uninstall avr-gcc and build from source with `brew install avr-gcc --build-from-source avr-gcc`.

I'm working on fixing the issue: the CI should have caught the broken linkage and fix it but it's not.

ladislas, 2020-01-05:
issue is now fixed with the latest revision bump, closing now. :)

closed 2020-01-05 by ladislas

mys721tx, 2020-01-05:
@ladislas Is there an ETA for the GCC 8 bottle?

ladislas, 2020-01-05:
@mys721tx working on it, still getting some errors with those.

ladislas, 2020-01-05:
@mys721tx should be good now! let me know if you have any issues.