osx-cross/homebrew-avr #176: New tiny (atxmega3 core) such as ATtiny817 seem to be missing

burnpanck, 2020-03-18:
Since AVR-GCC 8, the new tiny are supported: https://gcc.gnu.org/gcc-8/changes.html
If I do a `find  /usr/local/Cellar/avr-gcc -iname "*attiny817*"` I do get the following:
```
/usr/local/Cellar/avr-gcc/9.2.0_1/lib/avr-gcc/9/gcc/avr/9.2.0/device-specs/specs-attiny817
```
That specs file refers to a `crtattiny817.o` which seems to be missing.
If I instead search e.g. for the "ATxmega64a3u", I get
```
/usr/local/Cellar/avr-gcc/9.2.0_1/avr/lib/avrxmega4/libatxmega64a3u.a
/usr/local/Cellar/avr-gcc/9.2.0_1/avr/lib/avrxmega4/crtatxmega64a3u.o
/usr/local/Cellar/avr-gcc/9.2.0_1/lib/avr-gcc/9/gcc/avr/9.2.0/device-specs/specs-atxmega64a3u
```
In fact, `/usr/local/Cellar/avr-gcc/9.2.0_1/avr/lib/avrxmega3` seems to be completely missing.

ladislas, 2020-03-18:
In your example you're using avr-gcc 9.2.0.

Have you tried installing avr-gcc 8.x.x to see if it works?

```
brew install avr-gcc@8
```

burnpanck, 2020-03-18:
I had not, as support for new devices is usually cumulative; no reason to remove support for recent devices after they have just been added. Also, the presence of the specs file indicates that GCC thinks that the device is supported - the install is just broken in that it does not contain all the necessary files.

Anyway, I have now tried it with`avr-gcc@8` and I do get the exact same picture:
```
~ find  /usr/local/Cellar/avr-gcc@8/ -iname "*attiny817*"  
/usr/local/Cellar/avr-gcc@8//8.3.0_1/lib/avr-gcc/8/gcc/avr/8.3.0/device-specs/specs-attiny817
```
versus e.g.
```
~ find  /usr/local/Cellar/avr-gcc@8/ -iname "*atxmega64a3u*"
/usr/local/Cellar/avr-gcc@8//8.3.0_1/avr/lib/avrxmega4/libatxmega64a3u.a
/usr/local/Cellar/avr-gcc@8//8.3.0_1/avr/lib/avrxmega4/crtatxmega64a3u.o
/usr/local/Cellar/avr-gcc@8//8.3.0_1/lib/avr-gcc/8/gcc/avr/8.3.0/device-specs/specs-atxmega64a3u
```

burnpanck, 2020-03-18:
To be concrete, compilation of a simple empty loop `main()` with `avr-g++ -mmcu=attiny817 main.cpp -o main.elf ` fails as follows:
```
/usr/local/opt/avr-binutils/bin/avr-ld: cannot find crtattiny817.o: No such file or directory
/usr/local/opt/avr-binutils/bin/avr-ld: skipping incompatible /usr/local/Cellar/avr-gcc/9.2.0_1/lib/avr-gcc/9/gcc/avr/9.2.0/../../../../../../avr/lib/libm.a when searching for -lm
/usr/local/opt/avr-binutils/bin/avr-ld: cannot find -lm
/usr/local/opt/avr-binutils/bin/avr-ld: skipping incompatible /usr/local/Cellar/avr-gcc/9.2.0_1/lib/avr-gcc/9/gcc/avr/9.2.0/../../../../../../avr/lib/libm.a when searching for -lm
/usr/local/opt/avr-binutils/bin/avr-ld: cannot find -lm
/usr/local/opt/avr-binutils/bin/avr-ld: skipping incompatible /usr/local/Cellar/avr-gcc/9.2.0_1/lib/avr-gcc/9/gcc/avr/9.2.0/../../../../../../avr/lib/libc.a when searching for -lc
/usr/local/opt/avr-binutils/bin/avr-ld: cannot find -lc
/usr/local/opt/avr-binutils/bin/avr-ld: cannot find -lattiny817
collect2: error: ld returned 1 exit status
```
The same file compiles with e.g. `-mmcu=atxmega64a3u`.
Compilation against a random unsupported MCU fails differently:
```
~ avr-g++ -mmcu=atdoesnotexist1 main.cpp -o main.elf
avr-g++: fatal error: cannot read spec file 'device-specs/specs-atdoesnotexist1': No such file or directory
compilation terminated.
```

ladislas, 2020-03-19:
Do you have compilation errors when using `-mmcu=avrxmega3`? What does it say?

It might be the same kind of issue than the ones #105 fixed. I've also found this, maybe it can be helpful: 

> https://savannah.nongnu.org/patch/?9543

Could you try the patch and if it works make a PR? The one I referenced is a good example.

Feuerwerk, 2020-03-23:
Hi

If you need help i can support.

Greetings
Chris

ladislas, 2020-03-23:
Thanks a lot @Feuerwerk! :)

burnpanck, 2020-03-23:
Unfortunately, I'm quite tight on time these days. I probably won't be able to give it a try before April.

metropt, 2020-04-23:
I found the same today when I tried to use avr-gcc 5.5. With avr-gcc 8+ it is working.

burnpanck, 2020-04-23:
That  seems like a different issue to me. The new atxmega3 have really been added to avr-gcc around version 8 as you observed. The issue I encountered was in fact in `avr-libc`, where those devices are still not supported, except for a hanging pull-request. So I'm surprised that you say it does work for you with avr-gcc 8. Could you post the command-line you used there? I do note that you add include paths to some Atmel download related to DFPs - could it be possible that the missing crt files are found there?  Also, are you using the upstream avr-gcc or the Atmel distribution?

metropt, 2020-04-23:
The Atmel toolchain comes with GCC 5.4.0 https://www.microchip.com/mplab/avr-support/avr-and-arm-toolchains-c-compilers and I can run the above code. From what I understand you can give the compiler a device_specs with -B.

avr-gcc@8 works if I pass the -B option.

burnpanck, 2020-04-26:
Hey @metropt, thank you for your answer! Could you nonetheless please post the exact full command-line that did work for you?
I was able to download the device pack from atmel from packs.download.atmel.com. With that, I was able to compile my source with a command-line as follows:
```
avr-g++ -Batmel-DFP/gcc/dev/attiny817 -Iatmel-DFP/include  -Os -save-temps -g -gdwarf-3 -gstrict-dwarf -fno-exceptions   -mmcu=attiny817 -std=gnu++17 -o main.cpp.obj -c main.cpp
```
However, this does only compile (`-c` flag), so it will never show the link problems I posted above. If I try 
```
avr-g++ -Batmel-DFP/gcc/dev/attiny817 -Iatmel-DFP/include  -Os -save-temps -g -gdwarf-3 -gstrict-dwarf -fno-exceptions   -mmcu=attiny817 -std=gnu++17 -o main.elf main.cpp
```
instead, the missing `crtattiny817.o` and `libattiny817` are indeed found, but I still get `skipping incompatible ...` for `-lm` and `-lc`.

ladislas, 2020-04-26:
I would avoid at all costs mixing toolchains. Either go all the way with [Atmel toolchain](https://www.microchip.com/mplab/avr-support/avr-and-arm-toolchains-c-compilers) or with with this one.

metropt, 2020-04-26:
Thank you, with this toolchain I'm able to build the code only with GCC 9+, is that correct?

luqasz, 2020-05-13:
[GCC 10](https://gcc.gnu.org/gcc-10/changes.html) added some of them:
> ATtiny202, ATtiny204, ATtiny402, ATtiny404, ATtiny406, ATtiny804, ATtiny806, ATtiny807, ATtiny1604, ATtiny1606, ATtiny1607, ATmega808, ATmega809, ATmega1608, ATmega1609, ATmega3208, ATmega3209, ATmega4808, ATmega4809

burnpanck, 2020-05-13:
@luqasz: [GCC 8](https://gcc.gnu.org/gcc-8/changes.html) added the 1-series long before:

> The AVR port now supports the following XMEGA-like devices:
> ATtiny212, ATtiny214, ATtiny412, ATtiny414, ATtiny416, ATtiny417, ATtiny814, ATtiny816, ATtiny817, ATtiny1614, ATtiny1616, ATtiny1617, ATtiny3214, ATtiny3216, ATtiny3217
> The new devices are listed under -mmcu=avrxmega3.

The problem discussed in this issue is that [avr-libc](https://www.nongnu.org/avr-libc/), which is a separate project often bundled with avr-gcc (for example here in this `homebrew` package), was last released at version 2.0 four years ago and therefore does not support those new devices.

@ladislas has identified an [open patch to libc](https://savannah.nongnu.org/patch/?9543) adding that support, which, after my failed previous attempts to include here, have recently been updated.

Now, the chances might be much better - someone just needs to find the time for it :-).

ladislas, 2021-01-11:
Any news on this? I'll close the issue for now, feel free to reopen and submit a PR. :)

closed 2021-01-11 by ladislas