osx-cross/homebrew-avr #182: Linker fails on avr-gcc 8.4

fauxpark, 2020-03-23:
After updating to 8.4, avr-ld produces this error:
```
 | 
 | /usr/local/opt/avr-binutils/bin/avr-ld: cannot find crtatmega32u4.o: No such file or directory
 | /usr/local/opt/avr-binutils/bin/avr-ld: cannot find -latmega32u4
 | collect2: error: ld returned 1 exit status
 | 
```

and when building for AT90USB1286:
```
 | 
 | /usr/local/opt/avr-binutils/bin/avr-ld: cannot find crtat90usb1286.o: No such file or directory
 | /usr/local/opt/avr-binutils/bin/avr-ld: skipping incompatible /usr/local/Cellar/avr-gcc@8/8.4.0/lib/avr-gcc/8/gcc/avr/8.4.0/../../../../../../avr/lib/libm.a when searching for -lm
 | /usr/local/opt/avr-binutils/bin/avr-ld: cannot find -lm
 | /usr/local/opt/avr-binutils/bin/avr-ld: skipping incompatible /usr/local/Cellar/avr-gcc@8/8.4.0/lib/avr-gcc/8/gcc/avr/8.4.0/libgcc.a when searching for -lgcc
 | /usr/local/opt/avr-binutils/bin/avr-ld: cannot find -lgcc
 | /usr/local/opt/avr-binutils/bin/avr-ld: skipping incompatible /usr/local/Cellar/avr-gcc@8/8.4.0/lib/avr-gcc/8/gcc/avr/8.4.0/../../../../../../avr/lib/libm.a when searching for -lm
 | /usr/local/opt/avr-binutils/bin/avr-ld: cannot find -lm
 | /usr/local/opt/avr-binutils/bin/avr-ld: skipping incompatible /usr/local/Cellar/avr-gcc@8/8.4.0/lib/avr-gcc/8/gcc/avr/8.4.0/../../../../../../avr/lib/libc.a when searching for -lc
 | /usr/local/opt/avr-binutils/bin/avr-ld: cannot find -lc
 | /usr/local/opt/avr-binutils/bin/avr-ld: cannot find -lat90usb1286
 | collect2: error: ld returned 1 exit status
 | 
```

ladislas, 2020-03-23:
Yes, I've got the same issue, I'm investigating.

ladislas, 2020-03-23:
I've found the issue, it was the `--disable-multilib` option. I'll push a fix asap.

closed 2020-03-24 by ladislas

fauxpark, 2020-03-24:
Building succeeds again with the new bottles now - thanks for your hard work @ladislas :)

matsutakehoyo, 2020-03-25:
@fauxpark How did you  install with the new bottles? I tried reinstalling but I think it's just installing the same thing

myraz, 2020-03-29:
@matsutakehoyo I think i have the same problem. May I ask you how you erased your cashe?

matsutakehoyo, 2020-03-30:
@myraz when I tried to reinstall I got a prompt saying something like 
```
Reinstalling air-gcc@8 
==> Downloading https://homebrew...
Already downloaded: ... bottle.tar.gz
```
I just when to the directory that Already downloaded points to, found the downloaded file and manually deleted it. I think `brew cleanup` is supposed to remove these files, but on mine it is not. 

Hope this helps

ladislas, 2020-03-30:
@myraz 

```
brew uninstall avg-gcc OR avr-gcc@8
brew cleanup -s
brew install avr-gcc
```

myraz, 2020-03-30:
@matsutakehoyo @ladislas . Thank you very much for the quick help! At first it did not work but after unlinking my old avr-gcc and relinking, it now works!

```
brew unlink avr-gcc@8
brew link avr-gcc
```

ladislas, 2020-03-30:
Yes, you cannot have two different versions of gcc linked at the same time. 

avr-gcc will by default me linked and available in path. If you want another version linked, you must unlink the main one and manually link a @x version.

metropt, 2020-04-22:
I tried the above suggestions but I got no success compiling.

```
ze-mac:Downloads ze$ avr-gcc -mmcu=attiny202 -B ~/Downloads/Atmel.ATtiny_DFP.1.4.310/gcc/dev/attiny202 main.c -I ~/Downloads/Atmel.ATtiny_DFP.1.4.310/include/
In file included from main.c:30:
/usr/local/Cellar/avr-gcc/9.3.0/avr/include/util/delay.h:112:3: warning: #warning "Compiler optimizations disabled; functions from <util/delay.h> won't work as designed" [-Wcpp]
  112 | # warning "Compiler optimizations disabled; functions from <util/delay.h> won't work as designed"
      |   ^~~~~~~
/usr/local/opt/avr-binutils/bin/avr-ld: skipping incompatible /usr/local/Cellar/avr-gcc/9.3.0/lib/avr-gcc/9/gcc/avr/9.3.0/../../../../../../avr/lib/libm.a when searching for -lm
/usr/local/opt/avr-binutils/bin/avr-ld: cannot find -lm
/usr/local/opt/avr-binutils/bin/avr-ld: skipping incompatible /usr/local/Cellar/avr-gcc/9.3.0/lib/avr-gcc/9/gcc/avr/9.3.0/../../../../../../avr/lib/libc.a when searching for -lc
/usr/local/opt/avr-binutils/bin/avr-ld: cannot find -lc
collect2: error: ld returned 1 exit status
ze-mac:Downloads ze$ avr-gcc -v
Using built-in specs.
Reading specs from /usr/local/Cellar/avr-gcc/9.3.0/lib/avr-gcc/9/gcc/avr/9.3.0/device-specs/specs-avr2
COLLECT_GCC=avr-gcc
COLLECT_LTO_WRAPPER=/usr/local/Cellar/avr-gcc/9.3.0/libexec/gcc/avr/9.3.0/lto-wrapper
Target: avr
Configured with: ../configure --target=avr --prefix=/usr/local/Cellar/avr-gcc/9.3.0 --libdir=/usr/local/Cellar/avr-gcc/9.3.0/lib/avr-gcc/9 --enable-languages=c,c++ --with-ld=/usr/local/opt/avr-binutils/bin/avr-ld --with-as=/usr/local/opt/avr-binutils/bin/avr-as --disable-nls --disable-libssp --disable-shared --disable-threads --disable-libgomp --with-dwarf2 --with-avrlibc --with-pkgversion='Homebrew AVR GCC 9.3.0' --with-bugurl=https://github.com/osx-cross/homebrew-avr/issues
Thread model: single
gcc version 9.3.0 (Homebrew AVR GCC 9.3.0)
```

ladislas, 2020-04-22:
@metropt have you tried `brew remove avr-gcc` and then `brew install avr-gcc`?

metropt, 2020-04-22:
@ladislas `uninstall` and `install` didn't solve but `remove` and `install` solve the problem hmm

benwoodward, 2021-01-12:
I've tried all of the above and still experiencing this issue. 

```bash
❯ qmk setup benwoodward/qmk_firmware -b apple_fn_k320
Ψ Found qmk_firmware at /Users/ben/qmk_firmware.
Ψ QMK Doctor is checking your environment.
Ψ Detected macOS.
Ψ QMK home: /Users/ben/qmk_firmware
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Found avr-gcc version 8.4.0
Ψ Found avrdude version 6.3
Ψ Found dfu-util version 0.10
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go
❯ qmk compile -kb kbdfans/kbd75/rev2 -km default
Ψ Compiling keymap with make kbdfans/kbd75/rev2:default

QMK Firmware 0.11.43
Making kbdfans/kbd75/rev2 with keymap default

avr-gcc (Homebrew AVR GCC 8.4.0) 8.4.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Linking: .build/kbdfans_kbd75_rev2_default.elf                                                      [ERRORS]
 |
 | /usr/local/opt/avr-binutils/bin/avr-ld: cannot find crtatmega32u4.o: No such file or directory
 | /usr/local/opt/avr-binutils/bin/avr-ld: cannot find -latmega32u4
 | collect2: error: ld returned 1 exit status
 |
make[1]: *** [.build/kbdfans_kbd75_rev2_default.elf] Error 1
make: *** [kbdfans/kbd75/rev2:default] Error 1
Make finished with errors
```

ladislas, 2021-01-12:
yes, sorry for that! while working on the new bottle system, I took the opportunity to update the formula and follow what brew/core does. That's how I reintroduced the multilib thing...

it will soon be fixed.

w-A-L-L-e, 2021-01-12:
I'm trying this workaround (just as note both v9 and avr-gcc@10 also have this problem).
```
brew remove avr-gcc@8
brew uninstall avr-gcc
brew uninstall avr-binutils
wget https://raw.githubusercontent.com/osx-cross/homebrew-avr/3f86697315aad66d6213e158e0c0af88dd6c241f/Formula/avr-gcc%408.rb
brew install avr-gcc@8.rb
brew link --overwrite --force avr-gcc@8
```

ladislas, 2021-01-12:
This is fixed! 🎉 you can brew remove and brew install avr-gcc@8 again :)

w-A-L-L-e, 2021-01-15:
Good to know, for now the workaround worked for me and it takes a long time to recompile here so I'm leaving it. Any ideas when avr-gcc v9 and @10 will also get this fix?

ladislas, 2021-01-15:
everything is fixed, you need to `brew update && brew upgrade` first.

commit b0f5e1ed (2020-03-23): remove --disable-multilib to fix ld/linker error - closes #182
commit f232f806 (2020-03-23): remove --disable-multilib to fix ld/linker error - closes #182
commit 4caa4fc0 (2020-03-24): remove --disable-multilib to fix ld/linker error - closes #182

Closes #186.
commit c1c8167f (2020-03-24): avr-gcc@8 - remove --disable-multilib to fix ld/linker error - closes #182