osx-cross/homebrew-avr #191: Having issues compiling qmk firmwares.

obviouslygene, 2020-10-12:
MBP 16" 2019
macosx 10.15.7

I've tried reinstalling qmk environment.
Re-cloning qmk repo.

No go.

`❯ qmk compile
Ψ Compiling keymap with make basekeys/slice/rev1:via

QMK Firmware 0.10.33
Making basekeys/slice/rev1 with keymap via

avr-gcc (Homebrew AVR GCC 8.4.0) 8.4.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: quantum/quantum.c                                                                       during GIMPLE pass: *build_cgraph_edges
quantum/quantum.c: In function 'register_code16':
quantum/quantum.c:87:6: internal compiler error: Segmentation fault: 11
 void register_code16(uint16_t code) {
      ^~~~~~~~~~~~~~~
libbacktrace could not find executable to open
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://github.com/osx-cross/homebrew-avr/issues> for instructions.
 [ERRORS]
 |
 |
 |
make[1]: *** [.build/obj_basekeys_slice_rev1_via/quantum/quantum.o] Error 1
Make finished with errors
make: *** [basekeys/slice/rev1:via] Error 1`

ladislas, 2020-10-12:
Not sure I can help, you should report the issue to QMK.

cc @fauxpark

fauxpark, 2020-10-12:
I was actually the one who referred @obviouslygene here. I can't reproduce this myself, it's not a QMK issue.

ladislas, 2020-10-12:
@fauxpark ah sorry about that then!

I've never seen the issue and I'm not using avr-gcc as much as I used to in the past and I'm using avr-gcc@9.

@obviouslygene can you share a simple project with the issue that I can clone and test locally?

what's the full output of `brew doctor`? 

how did you install avr-gcc@8?

ladislas, 2020-10-12:
quick update: I've reinstalled avr-gcc@8, brew linked the binaries and rebuilt all my previous projects with it, I've got no errors and things run smoothly.

obviouslygene, 2020-10-13:
I've tried it with another mac and it works. I am not sure what is going on here either.

obviouslygene, 2020-10-13:
```❯ brew doctor
Your system is ready to brew.
❯ qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ Detected macOS.
Ψ QMK home: /Users/eliang/qmk_firmware
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Found avr-gcc version 8.4.0
Ψ Found avrdude version 6.3
Ψ Found dfu-util version 0.9
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go```

ladislas, 2020-11-17:
@obviouslygene any news? I'll close the issue for now, feel free to reopen.

closed 2020-11-17 by ladislas