osx-cross/homebrew-avr #193: OS X Big Sur issues

alxgarza, 2020-11-17:
After upgrading to OS 11 (Big Sur) and updating home-brew I get the following error:

lto1: fatal error: bytecode stream in file 'main.o' generated with LTO version 6.0 instead of the expected 8.1
compilation terminated.
lto-wrapper: fatal error: avr-gcc returned 1 exit status
compilation terminated.
/usr/local/opt/avr-binutils/bin/avr-ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make: *** [main.elf] Error 1

ladislas, 2020-11-17:
Thanks for reporting; Big Sur is not yet supported by homebrew. I'll wait for them to release the support version to look into this.

ladislas, 2020-12-14:
any news? I'm closing for now, feel free to reopen if needed.

closed 2020-12-14 by ladislas