osx-cross/homebrew-avr #198: Should we keep avr-gcc@4/5/6/7?

ladislas, 2021-01-11:
@osx-cross/team -- I'm in the process of moving to github actions to be able to fix all the bottle generation issues.

building gcc for all those versions takes an awful lot of time...

I think we should remove them and keep only the two latest, 9 & 10.

If needed, we can add them back for bottle building later-on.

If people really need them they can fork and it's now easier than ever to build your own bottles.

see: https://brew.sh/2020/11/18/homebrew-tap-with-bottles-uploaded-to-github-releases/

what do you guys think?

cc: @fauxpark @rleh

ladislas, 2021-01-11:
What I propose:

- I remove all the formulae
- I add them back one at a time starting with binutils and gcc@9
- when everything is up and running, I'll add @8 and @10
- then simavr and avarice
- then on a as requested basis, I'll add back the older versions if needed

larsimmisch, 2021-01-11:
Sounds good to me

fauxpark, 2021-01-11:
Definitely would prefer to keep 8.x around at least for the time being, 9 and 10 seem to compile somewhat larger binaries. Other than that, I agree Actions is the way to go.

ladislas, 2021-01-11:
Thanks all for the quick feedback :) 

I've started with #199 

@fauxpark: sure, I'll keep @8 as well once @9 has built successfully.

ladislas, 2021-01-11:
I've had issues with #199 because of auth and PAT but it seems to be working now. 

Feel free to test and report bugs. Especially with compilation (I haven't tried that). Binary diff would be a must though I don't expect any changes, but you never know.

ladislas, 2021-01-12:
The process is complete! 8, 9 and 10 are working fine and I've fixed the issues from #204.

closed 2021-01-12 by ladislas

fauxpark, 2021-01-12:
Thanks once again :)