osx-cross/homebrew-avr #200: dyld: Library not loaded: /usr/local/opt/isl/lib/libisl.22.dylib

kargeor, 2021-01-11:
```
dyld: Library not loaded: /usr/local/opt/isl/lib/libisl.22.dylib
  Referenced from: /usr/local/Cellar/avr-gcc/9.3.0/libexec/gcc/avr/9.3.0/cc1
  Reason: image not found
avr-gcc: internal compiler error: Abort trap: 6 signal terminated program cc1
Please submit a full bug report,
with preprocessed source if appropriate.
```

**MacOS Big Sur 11.0.1 (20B28)**

ladislas, 2021-01-11:
this should be fixed, can you `brew update` and `brew install avr-gcc` again?

kargeor, 2021-01-12:
after doing that now I get:
```
/usr/local/opt/avr-binutils/bin/avr-ld: cannot find crtatmega1284p.o: No such file or directory
/usr/local/opt/avr-binutils/bin/avr-ld: skipping incompatible /usr/local/Cellar/avr-gcc/9.3.0/lib/avr-gcc/9/gcc/avr/9.3.0/libgcc.a when searching for -lgcc
/usr/local/opt/avr-binutils/bin/avr-ld: cannot find -lgcc
/usr/local/opt/avr-binutils/bin/avr-ld: skipping incompatible /usr/local/Cellar/avr-gcc/9.3.0/lib/avr-gcc/9/gcc/avr/9.3.0/../../../../../../avr/lib/libm.a when searching for -lm
/usr/local/opt/avr-binutils/bin/avr-ld: cannot find -lm
/usr/local/opt/avr-binutils/bin/avr-ld: skipping incompatible /usr/local/Cellar/avr-gcc/9.3.0/lib/avr-gcc/9/gcc/avr/9.3.0/../../../../../../avr/lib/libc.a when searching for -lc
/usr/local/opt/avr-binutils/bin/avr-ld: cannot find -lc
/usr/local/opt/avr-binutils/bin/avr-ld: cannot find -latmega1284p
```

ladislas, 2021-01-12:
yes, everything is broken, see #204
will fix soon

salkinium, 2021-01-12:
This is still broken with the [latest 10.2.0 bottle](https://github.com/modm-io/modm/runs/1691710853):

```
Compiling C++·· build/release/modm/src/modm/utils/dummy.o
scons: building terminated because of errors.dyld: Library not loaded: /usr/local/opt/isl/lib/libisl.23.dylib
  Referenced from: /usr/local/Cellar/avr-gcc@10/10.2.0/libexec/gcc/avr/10.2.0/cc1plus
  Reason: image not found
avr-g++: internal compiler error: Abort trap: 6 signal terminated program cc1plus
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://github.com/osx-cross/homebrew-avr/issues> for instructions.
scons: *** [build/release/modm/src/modm/utils/dummy.o] Error 4
```

Clean install in GitHub Actions macOS 10.15 CI, we also still get the issue in #205.
```
brew tap osx-cross/avr
brew install avr-gcc@10
# Twice due to depends bullshit
brew install avr-gcc@10echo "/usr/local/opt/avr-gcc@10/bin" >> $GITHUB_PATH

avr-g++ --version

```

ladislas, 2021-01-13:
I don't recommend using 10.2.0. 

Can you try with 9? 

Also can you build from source?

Have you tried brew update && brew upgrade?

What is the output of brew doctor?

ladislas, 2021-01-13:
@salkinium I've just tested again, avr-gcc@10 works perfectly.

As said, try:

```
$ brew update && brew upgrade
$ brew cleanup -s
$ brew remove avr-gcc@10
$ brew install avr-gcc@10 && brew link --force avr-gcc@10
```

If it still doesn't work, you can:

```
$ brew install --build-from-source avr-gcc@10
$ brew link --force avr-gcc@10
```

Please also try `avr-gcc` (v9.3.0).

Also, this is wrong:

> `$ export HOMEBREW_NO_INSTALLED_DEPENDENTS_CHECK=1 # To avoid this bullshit`

You need it to install dependencies, or it will complain ISL is missing.

Finally, I don't understand this:

> `$ echo "/usr/local/opt/avr-gcc@10/bin" >> $GITHUB_PATH`

kargeor, 2021-01-13:
`brew install avr-gcc@10 && brew link --force avr-gcc@10` worked for me (MacOS Big Sur 11.0.1). Thanks.

salkinium, 2021-01-13:
I forgot to do `brew update` in our CI… 🤦‍♂️

This works fine now on the macOS 10.15 GitHub Actions CI, it also works fine locally on my 10.14 machine.
```
brew update
brew install avr-gcc@10
brew link --force avr-gcc@10
```

Thanks for your work!

ladislas, 2021-01-13:
Good to hear!

closed 2021-01-13 by ladislas