osx-cross/homebrew-avr #205: avr-gcc not in PATH on latest brew install

btashton, 2021-01-12:
This has been working great until today.  But with the latest brew install it seems that avr-gcc is not on the path

Note the line here
`==> Reinstalling 1 broken dependent from source:` Where it seem to be removing all of the symlinks.

```
+ brew tap osx-cross/avr
==> Tapping osx-cross/avr
Cloning into '/usr/local/Homebrew/Library/Taps/osx-cross/homebrew-avr'...
Tapped 4 formulae (37 files, 357.2KB).
+ brew install avr-gcc --verbose
==> Installing avr-gcc from osx-cross/avr
==> Downloading https://github.com/osx-cross/homebrew-avr/releases/download/avr-binutils-2.35.1/avr-binutils-2.35.1.catalina.bottle.tar.gz
==> Downloading from https://github-production-release-asset-2e65be.s3.amazonaws.com/30939945/0a804b00-5457-11eb-9d55-6d23e5d53055?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20210112%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210112T051751Z&X-Amz-Expires=300&X-Amz-Signature=748ab3ee7fa6acd5d4c0bee42d4c9e75ba61908815114b213efbdfb18126ce0d&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=30939945&response-content-disposition=attachment%3B%20filename%3Davr-binutils-2.35.1.catalina.bottle.tar.gz&response-content-type=application%2Foctet-stream
/usr/bin/curl --disable --globoff --show-error --user-agent Homebrew/2.6.1\ \(Macintosh\;\ Intel\ Mac\ OS\ X\ 10.15.7\)\ curl/7.64.1 --header Accept-Language:\ en --fail --silent --retry 3 --location --remote-time --continue-at 0 --output /Users/runner/work/incubator-nuttx-testing/incubator-nuttx-testing/sources/prebuilt/homebrew/downloads/007a35a3699672b503cd369e126af3ad4b37c8aaf7110841f30a733543d9e623--avr-binutils-2.35.1.catalina.bottle.tar.gz.incomplete https://github-production-release-asset-2e65be.s3.amazonaws.com/30939945/0a804b00-5457-11eb-9d55-6d23e5d53055\?X-Amz-Algorithm=AWS4-HMAC-SHA256\&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A\%2F20210112\%2Fus-east-1\%2Fs3\%2Faws4_request\&X-Amz-Date=20210112T051751Z\&X-Amz-Expires=300\&X-Amz-Signature=748ab3ee7fa6acd5d4c0bee42d4c9e75ba61908815114b213efbdfb18126ce0d\&X-Amz-SignedHeaders=host\&actor_id=0\&key_id=0\&repo_id=30939945\&response-content-disposition=attachment\%3B\%20filename\%3Davr-binutils-2.35.1.catalina.bottle.tar.gz\&response-content-type=application\%2Foctet-stream
==> Verifying checksum for '007a35a3699672b503cd369e126af3ad4b37c8aaf7110841f30a733543d9e623--avr-binutils-2.35.1.catalina.bottle.tar.gz'.
==> Downloading https://github.com/osx-cross/homebrew-avr/releases/download/avr-gcc-9.3.0/avr-gcc-9.3.0.catalina.bottle.2.tar.gz
==> Downloading from https://github-production-release-asset-2e65be.s3.amazonaws.com/30939945/018f7980-5457-11eb-890b-e22508470a41?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20210112%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210112T051751Z&X-Amz-Expires=300&X-Amz-Signature=4b22b5fa5815fd108834215e2a93a8ab9dfde457d79d2b9ce6d031aeaf54e271&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=30939945&response-content-disposition=attachment%3B%20filename%3Davr-gcc-9.3.0.catalina.bottle.2.tar.gz&response-content-type=application%2Foctet-stream
/usr/bin/curl --disable --globoff --show-error --user-agent Homebrew/2.6.1\ \(Macintosh\;\ Intel\ Mac\ OS\ X\ 10.15.7\)\ curl/7.64.1 --header Accept-Language:\ en --fail --silent --retry 3 --location --remote-time --continue-at 0 --output /Users/runner/work/incubator-nuttx-testing/incubator-nuttx-testing/sources/prebuilt/homebrew/downloads/36a148a4fec8b5202608c0ce59183d80c57fee3b7b9dd9422331a2ab948d8f14--avr-gcc-9.3.0.catalina.bottle.2.tar.gz.incomplete https://github-production-release-asset-2e65be.s3.amazonaws.com/30939945/018f7980-5457-11eb-890b-e22508470a41\?X-Amz-Algorithm=AWS4-HMAC-SHA256\&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A\%2F20210112\%2Fus-east-1\%2Fs3\%2Faws4_request\&X-Amz-Date=20210112T051751Z\&X-Amz-Expires=300\&X-Amz-Signature=4b22b5fa5815fd108834215e2a93a8ab9dfde457d79d2b9ce6d031aeaf54e271\&X-Amz-SignedHeaders=host\&actor_id=0\&key_id=0\&repo_id=30939945\&response-content-disposition=attachment\%3B\%20filename\%3Davr-gcc-9.3.0.catalina.bottle.2.tar.gz\&response-content-type=application\%2Foctet-stream
==> Verifying checksum for '36a148a4fec8b5202608c0ce59183d80c57fee3b7b9dd9422331a2ab948d8f14--avr-gcc-9.3.0.catalina.bottle.2.tar.gz'.
==> Installing dependencies for osx-cross/avr/avr-gcc: avr-binutils
==> Installing osx-cross/avr/avr-gcc dependency: avr-binutils
==> Pouring avr-binutils-2.35.1.catalina.bottle.tar.gz
tar xof /Users/runner/work/incubator-nuttx-testing/incubator-nuttx-testing/sources/prebuilt/homebrew/downloads/007a35a3699672b503cd369e126af3ad4b37c8aaf7110841f30a733543d9e623--avr-binutils-2.35.1.catalina.bottle.tar.gz -C /var/folders/24/8k48jl6d249_n_qfxwsl6xvm0000gn/T/d20210112-8203-1lt1ck0
cp -pR /var/folders/24/8k48jl6d249_n_qfxwsl6xvm0000gn/T/d20210112-8203-1lt1ck0/avr-binutils/. /usr/local/Cellar/avr-binutils
chmod -Rf +w /var/folders/24/8k48jl6d249_n_qfxwsl6xvm0000gn/T/d20210112-8203-1lt1ck0
==> Finishing up
ln -s ../Cellar/avr-binutils/2.35.1/bin/avr-addr2line avr-addr2line
ln -s ../Cellar/avr-binutils/2.35.1/bin/avr-ar avr-ar
ln -s ../Cellar/avr-binutils/2.35.1/bin/avr-as avr-as
ln -s ../Cellar/avr-binutils/2.35.1/bin/avr-c++filt avr-c++filt
ln -s ../Cellar/avr-binutils/2.35.1/bin/avr-elfedit avr-elfedit
ln -s ../Cellar/avr-binutils/2.35.1/bin/avr-gprof avr-gprof
ln -s ../Cellar/avr-binutils/2.35.1/bin/avr-ld avr-ld
ln -s ../Cellar/avr-binutils/2.35.1/bin/avr-ld.bfd avr-ld.bfd
ln -s ../Cellar/avr-binutils/2.35.1/bin/avr-nm avr-nm
ln -s ../Cellar/avr-binutils/2.35.1/bin/avr-objcopy avr-objcopy
ln -s ../Cellar/avr-binutils/2.35.1/bin/avr-objdump avr-objdump
ln -s ../Cellar/avr-binutils/2.35.1/bin/avr-ranlib avr-ranlib
ln -s ../Cellar/avr-binutils/2.35.1/bin/avr-readelf avr-readelf
ln -s ../Cellar/avr-binutils/2.35.1/bin/avr-size avr-size
ln -s ../Cellar/avr-binutils/2.35.1/bin/avr-strings avr-strings
ln -s ../Cellar/avr-binutils/2.35.1/bin/avr-strip avr-strip
ln -s ../../../Cellar/avr-binutils/2.35.1/share/man/man1/avr-addr2line.1 avr-addr2line.1
ln -s ../../../Cellar/avr-binutils/2.35.1/share/man/man1/avr-ar.1 avr-ar.1
ln -s ../../../Cellar/avr-binutils/2.35.1/share/man/man1/avr-as.1 avr-as.1
ln -s ../../../Cellar/avr-binutils/2.35.1/share/man/man1/avr-c++filt.1 avr-c++filt.1
ln -s ../../../Cellar/avr-binutils/2.35.1/share/man/man1/avr-dlltool.1 avr-dlltool.1
ln -s ../../../Cellar/avr-binutils/2.35.1/share/man/man1/avr-elfedit.1 avr-elfedit.1
ln -s ../../../Cellar/avr-binutils/2.35.1/share/man/man1/avr-gprof.1 avr-gprof.1
ln -s ../../../Cellar/avr-binutils/2.35.1/share/man/man1/avr-ld.1 avr-ld.1
ln -s ../../../Cellar/avr-binutils/2.35.1/share/man/man1/avr-nm.1 avr-nm.1
ln -s ../../../Cellar/avr-binutils/2.35.1/share/man/man1/avr-objcopy.1 avr-objcopy.1
ln -s ../../../Cellar/avr-binutils/2.35.1/share/man/man1/avr-objdump.1 avr-objdump.1
ln -s ../../../Cellar/avr-binutils/2.35.1/share/man/man1/avr-ranlib.1 avr-ranlib.1
ln -s ../../../Cellar/avr-binutils/2.35.1/share/man/man1/avr-readelf.1 avr-readelf.1
ln -s ../../../Cellar/avr-binutils/2.35.1/share/man/man1/avr-size.1 avr-size.1
ln -s ../../../Cellar/avr-binutils/2.35.1/share/man/man1/avr-strings.1 avr-strings.1
ln -s ../../../Cellar/avr-binutils/2.35.1/share/man/man1/avr-strip.1 avr-strip.1
ln -s ../../../Cellar/avr-binutils/2.35.1/share/man/man1/avr-windmc.1 avr-windmc.1
ln -s ../../../Cellar/avr-binutils/2.35.1/share/man/man1/avr-windres.1 avr-windres.1
/usr/bin/sandbox-exec -f /private/tmp/homebrew20210112-8401-197b7ps.sb nice ruby -W0 -I $LOAD_PATH -- /usr/local/Homebrew/Library/Homebrew/postinstall.rb /usr/local/Homebrew/Library/Taps/osx-cross/homebrew-avr/Formula/avr-binutils.rb
==> Summary
🍺  /usr/local/Cellar/avr-binutils/2.35.1: 158 files, 22MB
==> Installing osx-cross/avr/avr-gcc
==> Pouring avr-gcc-9.3.0.catalina.bottle.2.tar.gz
tar xof /Users/runner/work/incubator-nuttx-testing/incubator-nuttx-testing/sources/prebuilt/homebrew/downloads/36a148a4fec8b5202608c0ce59183d80c57fee3b7b9dd9422331a2ab948d8f14--avr-gcc-9.3.0.catalina.bottle.2.tar.gz -C /var/folders/24/8k48jl6d249_n_qfxwsl6xvm0000gn/T/d20210112-8203-1ecjhvo
cp -pR /var/folders/24/8k48jl6d249_n_qfxwsl6xvm0000gn/T/d20210112-8203-1ecjhvo/avr-gcc/. /usr/local/Cellar/avr-gcc
chmod -Rf +w /var/folders/24/8k48jl6d249_n_qfxwsl6xvm0000gn/T/d20210112-8203-1ecjhvo
==> Finishing up
[... 42 lines cut ...]
rm /usr/local/share/man/man1/avr-g++.1
rm /usr/local/share/man/man1/avr-gcc.1
rm /usr/local/share/man/man1/avr-gcov-dump.1
[... cut ...]
/usr/local/Cellar/avr-gcc/9.3.0.reinstall (1,713 files, 94.2MB)
  Poured from bottle on 2021-01-12 at 05:17:57
From: https://github.com/osx-cross/homebrew-avr/blob/HEAD/Formula/avr-gcc.rb
==> Dependencies
Build: autoconf, automake
Required: avr-binutils, gmp, isl, libmpc, mpfr
==> Requirements
Required: x86_64 architecture
==> Options
--with-ATMega168pbSupport
	Add ATMega168pb Support to avr-gcc
--HEAD
	Install HEAD version
+ avr-gcc --version
./cibuild.sh: line 228: avr-gcc: command not found
```

btashton, 2021-01-12:
Running `brew install` twice results in it installing, but then I hit the issue in #200

ladislas, 2021-01-12:
yes, everything is broken, see #204
will fix soon

ladislas, 2021-01-12:
Fix is available, can you brew uninstall and brew install again?

ladislas, 2021-01-12:
avr-gcc@8/9/10 are now available and working fine. I'll close the issue. Feel free to reopen or create a new one if needed.

yamt, 2021-01-15:
@ladislas it doesn't seem working for our case.

```+ brew tap osx-cross/avr
==> Tapping osx-cross/avr
Cloning into '/usr/local/Homebrew/Library/Taps/osx-cross/homebrew-avr'...
Tapped 5 formulae (38 files, 372.7KB).
+ brew install avr-gcc
==> Installing avr-gcc from osx-cross/avr
==> Downloading https://github.com/osx-cross/homebrew-avr/releases/download/avr-binutils-2.35.1/avr-binutils-2.35.1.catalina.bottle.tar.gz
==> Downloading from https://github-production-release-asset-2e65be.s3.amazonaws.com/30939945/0a804b00-5457-11eb-9d55-6d23e5d53055?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20210114%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210114T014049Z&X-Amz-Expires=300&X-Amz-Signature=1e6293055fd3b03415131480460a8ad897e017aeb9efc033f1430232e87e743d&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=30939945&response-content-disposition=attachment%3B%20filename%3Davr-binutils-2.35.1.catalina.bottle.tar.gz&response-content-type=application%2Foctet-stream
==> Downloading https://github.com/osx-cross/homebrew-avr/releases/download/avr-gcc-9.3.0/avr-gcc-9.3.0.catalina.bottle.3.tar.gz
==> Downloading from https://github-production-release-asset-2e65be.s3.amazonaws.com/30939945/19551500-54cc-11eb-8855-d190dec8b7e2?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20210114%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210114T014049Z&X-Amz-Expires=300&X-Amz-Signature=bfa0b2ffd2f6476ff0021644a51cb12013c1590897527be7abeffad4d92ca383&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=30939945&response-content-disposition=attachment%3B%20filename%3Davr-gcc-9.3.0.catalina.bottle.3.tar.gz&response-content-type=application%2Foctet-stream
==> Installing dependencies for osx-cross/avr/avr-gcc: avr-binutils
==> Installing osx-cross/avr/avr-gcc dependency: avr-binutils
==> Pouring avr-binutils-2.35.1.catalina.bottle.tar.gz
🍺  /usr/local/Cellar/avr-binutils/2.35.1: 158 files, 22MB
==> Installing osx-cross/avr/avr-gcc
==> Pouring avr-gcc-9.3.0.catalina.bottle.3.tar.gz
🍺  /usr/local/Cellar/avr-gcc/9.3.0: 1,749 files, 206.9MB
==> No outdated dependents to upgrade!
==> Checking for dependents of upgraded formulae...
==> Reinstalling 1 broken dependent from source:
osx-cross/avr/avr-gcc
```

the full log is here: https://github.com/apache/incubator-nuttx-testing/pull/88/checks?check_run_id=1699065257#step:6:545

yamt, 2021-01-15:
it might be macOS version specific as i couldn't reproduce the issue on my local environment. (mojave)

salkinium, 2021-01-15:
Had the same issue with GH Actions, you need to do `brew update` before installing.

yamt, 2021-01-15:
> Had the same issue with GH Actions, you need to do `brew update` before installing.

thank you. let me try.

ladislas, 2021-01-15:
> it might be macOS version specific as i couldn't reproduce the issue on my local environment. (mojave)

We don't provide Mojave bottles anymore, as the CI system only provide Catalina and BigSur VM.

You'll need to build from source and maybe tweak the formula a little. I'm won't be able to help you here as I don't have a machine running Mojave.

Can you upgrade to a more recent macOS version?

Please also provide me with the output of brew doctor.

btashton, 2021-01-15:
@ladislas this is Catalina that it fails on. It looks like brew upgrade might need fixing it but that is causing an unrelated issue.

ladislas, 2021-01-15:
In this case you can always build from source but it will take between 20minutes and several hours depending on your machine.

I'll close as it is unrelated to the tap. I recommend forcing your CI to brew update/brew upgrade often so that you can spot this kind of issues and fix them quickly

closed 2021-01-15 by ladislas

btashton, 2021-01-15:
It's a bug in the GitHub Actions runner environment unfortunately impacting anything that uses python 3. Brew upgrade adds a non trivial amount of time to the build so we normally rely on GitHub to keep the base environment up-to-date.

ladislas, 2021-01-15:
> Brew upgrade adds a non trivial amount of time to the build so we normally rely on GitHub to keep the base environment up-to-date.

Yeah that's a pain in the neck... I found out that brew update and then brew upgrade only the formula you need can help, but if the CI env is really out of date it will break other things too.

For another project, I recycled an old macbook pro 13 to run our CI -- it's really easy to add self-hosted runners to Github and is a game changer 

Hope you'll find a fix soon.

Hid-K, 2022-07-07:
I had the same issue. Brew cleanup showed this:
```
Warning: Some installed formulae are not readable:
  avr-binutils: Invalid bottle tag symbol

  avr-gdb: Invalid bottle tag symbol
```
and this:
```
Warning: Some taps are not on the default git origin branch and may not receive
updates. If this is a surprise to you, check out the default branch with:
  git -C $(brew --repo osx-cross/avr) checkout main

Warning: Homebrew's "sbin" was not found in your PATH but you have installed
formulae that put executables in /usr/local/sbin.
Consider setting your PATH for example like so:
  echo 'export PATH="/usr/local/sbin:$PATH"' >> /Users/USERNAME/.bash_profile
```

# For me solution was:
```
 git -C $(brew --repo osx-cross/avr) checkout main && echo 'export PATH="/usr/local/sbin:$PATH"' >> /Users/USERNAME/.bash_profile
```