osx-cross/homebrew-avr #207: avr-gcc@8 has no atmega32u4 libraries

w-A-L-L-e, 2021-01-12:
After doing a brew update, I ran into an issue with avr-gcc vs 9.3.0 it doesnt compile a library called micro-ecc anymore.
The error is string too long in micro-ecc/asm_avr.inc which is basically an unrolled asm function (and therefor a long string with asm instructions). 

Long story short. So I tried installing avr-gcc@8 and then linked it as default. Now it does compile all my c and asm files for my project. But when it starts linking I get these missing library errors:
```
/usr/local/opt/avr-binutils/bin/avr-ld: cannot find crtatmega32u4.o: No such file or directory
/usr/local/opt/avr-binutils/bin/avr-ld: cannot find -latmega32u4
```

ladislas, 2021-01-12:
yes, everything is broken, see #204
will fix soon

ladislas, 2021-01-12:
Can you brew remove avr-gcc@9 and brew install again? it should be fixed :)

ladislas, 2021-01-12:
avr-gcc@8/9/10 are now available and working fine. I'll close the issue. Feel free to reopen or create a new one if needed.

closed 2021-01-12 by ladislas