osx-cross/homebrew-avr #214: avr-project in homebrew-avr?

GGorAA, 2021-01-18:
As in title

ladislas, 2021-01-18:
Can you be a little more precise?

GGorAA, 2021-01-19:
So, I think a project called x-avr uses avr-project tool to make Xcode projects for AVR. But when I uninstalled CrossPack and installed homebrew-avr, there was no avr-project. And I think that was the cause of x-avr failing to generate a proper project

ladislas, 2021-01-19:
ah! I already had this kind of question in the past. See #131 & #132.

avr-project and osx-cross/avr are not related in any way, I've never ever used avr-project, nor crosspack, nor xcode templates for avr in my life, so I can't really help you.

If you think they or documentation could be added to this repo, you can open a PR and we'll discuss it.

GGorAA, 2021-01-19:
I understand, thanks for replying :D

closed 2021-01-19 by GGorAA