osx-cross/homebrew-avr #215: Compiling for AVR64DA48

fruchtkeks, 2021-01-26:
Hi,

I'm trying to compile a C firmware for the AVR64DA48. The avr-gcc fails with the following error:

> avr-gcc -mmcu=avr64da28
avr-gcc: fatal error: cannot read spec file 'device-specs/specs-avr64da28': No such file or directory
compilation terminated.

Since the AVR64DA48 is relatively new (mid 2020), it seems that device-specs are missing.
I downloaded the AVR-Dx Series Device Support (https://packs.download.microchip.com) and tried compiling with the following command:

> avr-gcc -mmcu=avr64da48 -B../1.6.76/gcc/dev/avr64da48 -I../1.6.76/include main.c
/usr/local/opt/avr-binutils/bin/avr-ld: address 0x806000 of a.out section `.data' is not within region `data'
/usr/local/opt/avr-binutils/bin/avr-ld: address 0x806000 of a.out section `.data' is not within region `data'
collect2: error: ld returned 1 exit status

I think I'm still missing something here. Therefore my question:

Where do the device-specs come from and how are they included in avr-gcc?
Has had anyone luck with compiling for the AD Series?

ladislas, 2021-01-26:
it's the same as #176 -- a patch to avr-libc must be provided.

ladislas, 2021-02-04:
@fruchtkeks any news?

I'll close for now, feel free to open a PR with the changes for review.

closed 2021-02-04 by ladislas