osx-cross/homebrew-avr #216: avr-gdb: Symbol format `elf32-avr' unknown.

xhero, 2021-01-26:
Hi,
I'm getting a weird issue with avr-gdb on macOS 11.1. I just installed avr-gdb via brew which works correctly, but then when I try to load one of my elf files I get:

```I'm sorry, Dave, I can't do that.  Symbol format `elf32-avr' unknown.```

It seems the bfd definitions are not present? After a bit of digging I could not figure out why they are not there, as if I understand correctly they should be supplied by avr-binutils.

Any pointers to fix this?
Thanks!

ladislas, 2021-01-26:
Have you installed avt-binutils

xhero, 2021-01-26:
yes!

ladislas, 2021-01-26:
Hmm, I don't know what could cause that.

Could you edit the formula by removing the `--with-python` option here https://github.com/osx-cross/homebrew-avr/blob/master/Formula/avr-gdb.rb#L36 and then building from source (it's not too long)?

It's the only thing that changed besides the version.

xhero, 2021-01-27:
It seems a problem on macOS using clang:

https://unix.stackexchange.com/questions/624486/gdb-installed-from-homebrew-on-macos-big-sur-fails-to-load-symbols-from-elf32-i3

and

https://www.cygwin.com/bugzilla/show_bug.cgi?id=26949

Unfortunately none of the patches worked for me. It did work using gcc 10 from homebrew:

```bash
brew install gcc
HOMEBREW_CC=gcc-10 HOMEBREW_CXX=g++-10 brew reinstall --build-from-source avr-gdb
```

And then the problem disappears.

ladislas, 2021-01-27:
I'm glad you fixed the issue. But depending on `gcc` is not a solution for us.

Did you try this one? https://unix.stackexchange.com/a/624673

> The fix is just to add #include <string.h> to the top of elf-bfd.h and GDB will correctly load ELF binaries.

xhero, 2021-01-27:
yes, it makes no difference unfortunately. I can try to debug it a bit more, it seems that on some systems adding the include works and in others no.

ladislas, 2021-01-27:
If you have time, that would be great. I confess I haven't used it in month and I don't have any projects right now that would need it, so it's not really on my priority list.

But your solution might help others! 

Seem like the issue is really with gdb and not just avr-gdb.

ladislas, 2021-01-27:
Have you tried mixing the patches?

ladislas, 2021-02-04:
@xhero any news?

ladislas, 2021-02-11:
@xhero I'll close for now, feel free to reopen with new info if you have.

closed 2021-02-11 by ladislas

danapplegate, 2021-07-30:
Hi @ladislas! Just wanted to let you know that in the [other thread](https://github.com/osx-cross/homebrew-avr/issues/251) on this issue, they've also identified that adding `#include <string.h>` to `bfd/bfd.h` is a fix, and it seems that #255 includes patches to `avr-binutils` and `avr-gdb` that do just that. After uninstalling `avr-gcc` and `avr-gdb` and adding the patches from that PR, I was able to successfully `brew reinstall --build-from-source avr-binutils` then `brew install avr-gcc@11 avr-gdb`, and it seems to have resolved the problem!

Hope that PR can get merged soon and the distributions rebuilt, I imagine other people are running into this. Thanks for your work on this!