osx-cross/homebrew-avr #217: __flash address space qualifier missing

henryorton, 2021-02-01:
Hi,

I installed this on mac os 10.13.6 with the hope of using the __flash address qualifier (introduced in place of PROGMEM for avr-gcc 4.7 onwards).

For a program:

    #include <avr/io.h>
    const __flash int i = 5;
    int main(void)
    {
      return 0;
    }

I get an error with avr-gcc (v9.3.0):

    main.c:6:14: error: expected ';' before 'int'

Any ideas why __flash is not recognised?

ladislas, 2021-02-01:
No, I'm sorry. And this is not exactly the right place as this repo is only for the brew formula.

Take a look here:

- https://gcc.gnu.org/onlinedocs/gcc-6.1.0/gcc/Named-Address-Spaces.html
- https://github.com/jnk0le/AVR-UART-lib/issues/6
- https://www.avrfreaks.net/forum/random-output-when-reading-flash-lcd

Are you using avr-g++ by any chance? `__flash` is only for avr-gcc

closed 2021-02-01 by ladislas

henryorton, 2021-02-01:
Hey sorry to waste your time. Turns out I had the -std=c99 argument on
avr-gcc and this was not allowing __flash.
Cheers for the up to date toolchain on homebrew. It's a big improvement
over the outdated crosspack for mac.

ladislas, 2021-02-01:
Glad you found the issue! It might help others in the future :)