osx-cross/homebrew-avr #220: M1 Mac Support

Grippy98, 2021-02-04:
If I try to build from source I get:

avr-gcc: The x86_64 architecture is required for this software.

Any idea if this will/can be easily ported to Apple Silicon? (It's ARM64 after all)

Any tips would be great, would love to help with it if you can give me some pointers. Thanks!

Grippy98, 2021-02-04:
Looks like removing the "x86_x64" platform requirement works up until this point:

===> make install
🍺  /opt/homebrew/Cellar/avr-binutils/2.35.1: 158 files, 12.5MB, built in 51 seconds
gripppy@Andreis-Air Formula % brew install --build-from-source avr-gcc.rb
==> Downloading https://download.savannah.gnu.org/releases/avr-libc/avr-libc-2.0.0.tar.bz2
==> Downloading from https://bigsearcher.com/mirrors/nongnu/avr-libc/avr-libc-2.0.0.tar.bz2
######################################################################## 100.0%
==> Downloading https://ftp.gnu.org/gnu/gcc/gcc-9.3.0/gcc-9.3.0.tar.xz
######################################################################## 100.0%
==> ../configure --target=avr --prefix=/opt/homebrew/Cellar/avr-gcc/9.3.0_1 --libdir=/opt/homebrew/Cellar/avr-gcc/9.3.0_1/lib/avr-gcc/9 --enable-languages=c,c++ --with-ld=/opt/homebrew/opt/avr-binutils/bin/avr-ld --with-as=/opt/homebrew/o
==> make BOOT_LDFLAGS=-Wl,-headerpad_max_install_names
Last 15 lines from /Users/gripppy/Library/Logs/Homebrew/avr-gcc/02.make:
		lto/lto-lang.o lto/lto.o lto/lto-object.o attribs.o lto/lto-partition.o lto/lto-symtab.o libbackend.a main.o libcommon-target.a libcommon.a ../libcpp/libcpp.a ../libdecnumber/libdecnumber.a -lisl -lmpc -lmpfr -lgmp   -lz libcommon.a ../libcpp/libcpp.a  -liconv ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a
clang++ -fno-PIE -c  -DIN_GCC_FRONTEND -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-strict-aliasing -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../../gcc -I../../gcc/. -I../../gcc/../include -I../../gcc/../libcpp/include  -I../../gcc/../libdecnumber -I../../gcc/../libdecnumber/dpd -I../libdecnumber -I../../gcc/../libbacktrace   -o cc1-checksum.o -MT cc1-checksum.o -MMD -MP -MF ./.deps/cc1-checksum.TPo cc1-checksum.c
clang++ -fno-PIE -c  -DIN_GCC_FRONTEND -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-strict-aliasing -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../../gcc -I../../gcc/. -I../../gcc/../include -I../../gcc/../libcpp/include  -I../../gcc/../libdecnumber -I../../gcc/../libdecnumber/dpd -I../libdecnumber -I../../gcc/../libbacktrace   -o cc1plus-checksum.o -MT cc1plus-checksum.o -MMD -MP -MF ./.deps/cc1plus-checksum.TPo cc1plus-checksum.c
Undefined symbols for architecture arm64:
  "_host_hooks", referenced from:
      gt_pch_save(__sFILE*) in libbackend.a(ggc-common.o)
      gt_pch_restore(__sFILE*) in libbackend.a(ggc-common.o)
      toplev::main(int, char**) in libbackend.a(toplev.o)
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lto1] Error 1
make[2]: *** Waiting for unfinished jobs....
rm gcc.pod
make[1]: *** [all-gcc] Error 2
make: *** [all] Error 2

**The part that actually fails is this:**

Undefined symbols for architecture arm64:
  "_host_hooks", referenced from:
      gt_pch_save(__sFILE*) in libbackend.a(ggc-common.o)
      gt_pch_restore(__sFILE*) in libbackend.a(ggc-common.o)
      toplev::main(int, char**) in libbackend.a(toplev.o)
ld: symbol(s) not found for architecture arm64

Looks like an issue with the linker right?

ladislas, 2021-02-04:
GCC is not supported on M1 Mac, see:

- https://github.com/Homebrew/brew/issues/7857
- https://github.com/Homebrew/brew/issues/7857#issuecomment-687068494

So for the moment it's not possible to run avr-gcc on Apple M1. From what I read, even Rosetta 2 doesn't seem to do the tricK.

Grippy98, 2021-02-04:
https://github.com/riscv/homebrew-riscv/issues/47#issuecomment-772306695 

It does look like others have gotten RISCV-GCC to work without PCH.

ladislas, 2021-02-04:
Please open à PR and I'll be happy to review

jknotzke, 2021-02-04:
I am confused by the comment that gcc is not supported on the M1:

If I do the following:

```
bin file gcc
gcc: Mach-O universal binary with 2 architectures: [x86_64:Mach-O 64-bit executable x86_64] [arm64e:Mach-O 64-bit executable arm64e]
gcc (for architecture x86_64):	Mach-O 64-bit executable x86_64
gcc (for architecture arm64e):	Mach-O 64-bit executable arm64e
```

I must be missing something..

ladislas, 2021-02-05:
The first one is not gcc, its clang. 
The second one I don't know. 

I have zero way of testing if avt-gcc would work and currently day job is taking a lot of space. 

As you own an M1 Mac, try the different solutions and if it works submit a PR that I'll gladly review.

ladislas, 2021-02-05:
You can look at the gcc.rb formula in brew/core to see how they do it. They use a different url, a fork for the M1. But there is absolutely no guarantee that it will work, work with avr or create good binaries.

ladislas, 2021-02-05:
Here:

https://github.com/Homebrew/homebrew-core/blob/fcad0de971d82981d60bae5d5fade422a9dfc12d/Formula/gcc.rb#L4-L15

jknotzke, 2021-02-05:
In 

`/opt/homebrew/Library/Taps/osx-cross/homebrew-avr/Formula`

I removed all dependencies to x86_64. The build ends with:

```
Already downloaded: /Users/justin/Library/Caches/Homebrew/downloads/e0f07e5642c8cb09023d4947e24bdde272b63df8ca8bbe934bd7b75ff15f4e89--qmk-0.0.39.tar.gz
==> Installing dependencies for qmk/qmk/qmk: libelf, libusb, libusb-compat, libftdi0, libhid, avrdude, bootloadhid, clang-format, dfu-programmer, dfu-util, osx-cross/arm/arm-gcc-bin@8, avr-binutils, gmp, isl, mpfr, libmpc, osx-cross/avr/avr-gcc@8, gdbm, openssl@1.1, readline, sqlite, tcl-tk, xz, python and teensy_loader_cli
==> Installing qmk/qmk/qmk dependency: libelf
==> autoreconf -fvi
==> ./configure --prefix=/opt/homebrew/Cellar/libelf/0.8.13_1 --disable-compat
Last 15 lines from /Users/justin/Library/Logs/Homebrew/libelf/02.configure:
checking the coffee machine... empty - operator may not work as expected
checking whether 64-bit ELF support is sufficient... yes
checking whether to include 64-bit support... yes
checking whether versioning support is sufficient... yes
checking whether to include versioning support... yes
checking whether NLS is requested... yes
checking for dgettext... no
checking for catgets... yes
checking for gencat... /usr/bin/gencat
checking for gmsgfmt... msgfmt
checking for xgettext... xgettext
checking for catalogs to be installed... de
checking for gettext in -lintl... no
checking build system type... configure: error: /bin/sh ./config.sub -apple-darwin20.3.0 failed
configure: WARNING: cache variable ac_cv_build contains a newline

Do not report this issue to Homebrew/brew or Homebrew/core!

These open issues may also help:
libelf: revision for ARM https://github.com/Homebrew/homebrew-core/pull/68762
```

Getting closer.. (I think)

Grippy98, 2021-02-05:
"checking build system type... configure: error: /bin/sh ./config.sub -apple-darwin20.3.0 failed"

Got architecture build flag is usually something like "aarch64-apple-darwin" or "arm64-apple-darwin".

I'm stuck with day-job things for a bit but I'd look at what ./config.sub it downloads in that first line. (just an idea, and see what the diff is with something like the homebrew-riscv guys are doing)

ladislas, 2021-02-11:
@jknotzke @Grippy98 any news?

jknotzke, 2021-02-11:
I am stuck building libelf. I placed a comment https://github.com/Homebrew/homebrew-core/pull/68762 and so far no replies. Someone did post a patch, which did not work. I think the bottleneck is there (no pun). 

  How can I locate the source code for libelf that is being used to build for brew ? Also, how can I find the configure flags that are being passed ? If I can get libelf to build, I think we are almost there.

jknotzke, 2021-02-11:
Here is the config.log for libelf if that helps..
[config.log](https://github.com/osx-cross/homebrew-avr/files/5966131/config.log)

jknotzke, 2021-02-11:
I found the source code being used for libelf in brew. I downloaded it, and if I do the following:

 echo 'echo arm-apple-darwin' > config.sub

 and then run: ./configure --prefix=/opt/homebrew/Cellar/libelf/0.8.13_1 --disable-compat

It configures and I can run make with no errors..

But I haven't been able to locate where brew installs the source code for libelf.. If I could, I could test to see if this gets me any further..

ladislas, 2021-02-11:
> avr-gcc works great under Rosetta (it still compiles binaries significantly faster than my Windows desktop somehow) but I haven't had time to try native, was going to try to forge ahead over the weekend. Not sure if @jknotzke has had any progress.

@Grippy98 well that should be sufficient then! GCC is not officially supported for the M1, so I don't think it's a good idea to waste time hacking things when rosetta can do the trick.

Did you do anything to make it work with rosetta? 

@jknotzke why not use rosetta?

jknotzke, 2021-02-11:
I have it working with Rosetta.. But I suspect this solution is only viable for a while. It would obviously be better to get it working natively.

ladislas, 2021-02-11:
Rosetta 2 is here to stay for some time. GCC will bring native support for the M1.

That being said, avr will be removed from GCC 11, so we're kind of stuck here :(

larsimmisch-adobe, 2021-02-11:
What will be next? clang?

(not that I think we couldn't compile gcc natively for ARM if we had to)

ladislas, 2021-02-11:
Some guys are working on a port, but it doesn't look easy.

- https://www.bountysource.com/issues/84630749-avr-convert-the-backend-to-mode_cc-so-it-can-be-kept-in-future-releases
- https://gcc.gnu.org/bugzilla/show_bug.cgi?id=92729
- https://www.avrfreaks.net/forum/avr-gcc-and-avr-g-are-deprecated-now?page=all
- https://www.reddit.com/r/gcc/comments/dq3duj/gcc_11_avr_support/

larsimmisch-adobe, 2021-02-11:
Oh, I see, thanks - the avr backend needs porting for gcc 11. That's a whole different kettle of fish :(

ladislas, 2021-02-11:
I feel sorry for the guy working on it, the gcc-patch submission process seems like hell on earth...

jknotzke, 2021-02-11:
In case someone finds this thread and wants to be able to install qmk using Rosetta here are the steps I followed:

First, install brew but as X86: 
arch -x86_64 /bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install.sh)"

Then install qmk
arch -x86_64 brew install qmk/qmk/qmk

Then for qmk setup:
arch -x86_64 qmk setup -H /where/you/want

ladislas, 2021-02-19:
As there's not much we can do about it now and this is all too experimental to actually be merged into master, I'll close the issue for now.

We can discuss it again when GCC 11 comes out with hopefully avr support.

closed 2021-02-19 by ladislas

invigorate, 2021-12-17:
> We can discuss it again when GCC 11 comes out with hopefully avr support.

https://github.com/osx-cross/homebrew-avr/pull/248

If I'm reading this correctly, avr-gcc added support for M1 in June, and avr-gcc@11 is out.