osx-cross/homebrew-avr #221: Request for adding avra(assembly compiler) to the formulae

GGorAA, 2021-02-04:
As in the title, can you add `avra`, the AVR assembly compiler, to the `homebrew-avr` formulae?

ladislas, 2021-02-04:
I don't know what it is but I'll be happy to review a PR.

GGorAA, 2021-02-05:
Basically it is `avr-gcc`, but for Assembly, called `avra` from AVR Assembly. I'll create a PR with `avra` included :D

ladislas, 2021-03-06:
@GGorAA any news?

GGorAA, 2021-03-06:
Ohh, I'm sorry, I completely forgot about it :D
When I will find time I will create a PR :)

ladislas, 2021-03-09:
You don't have to, the CI will do it for you. I just need the PR.

GGorAA, 2021-03-14:
I mean, how to add avra compiler to the formulae, I have no idea where to place it

GGorAA, 2021-03-15:
Thanks for the explanation!

bartekpacia, 2021-05-27:
I'll try to do this:)

ladislas, 2021-10-05:
closing as won't fix for the moment. Feel free to reopen #235 in the future if needed.

closed 2021-10-05 by ladislas