osx-cross/homebrew-avr #223: Standard version of compiler is 5.4.0

fuxelius, 2021-02-05:
Hi!
The standard compiler used by professionals is actually AVR-GCC 5.4.0 which is the one provided by both Microchip and by Ubuntu 20.04. This one is seen as the the one producing the best code and the official number will probably be fixed at 5.4.0 forever since development by GCC on AVR8 has stopped. 

It would therefore be superhandy to have the possibility to have AVR-GCC 5.4.0 and AVR-GNU Linker 5.4.0 for the MacOS environment. 

Thanx and all the Best!
Hans-Henrik Fuxelius

ladislas, 2021-02-05:
Can you check out PR #225 from branch formula/avr-gcc@5/add-formula and tell me if it works locally.

If not, you can use https://github.com/Homebrew/homebrew-core/blob/master/Formula/gcc@5.rb to try and find a solution.

closed 2021-02-07 by osxCrossTestBot

ladislas, 2021-02-07:
@fuxelius avr-gcc@5 is not available, you can install with:

```
brew install avr-gcc@5
```

Open a new issue if things don't work the way they should.

commit b0bb47ae (2021-02-06): avr-gcc@5 - add formula

closes #223
commit 7e13f54a (2021-02-07): avr-gcc@5 - add formula

closes #223
commit c94c620e (2021-02-07): avr-gcc@5 - add formula

closes #223

Closes #225.