osx-cross/homebrew-avr #229: libbacktrace could not find executable to open

paradajz, 2021-02-10:
Running on macOS Catalina. Not sure what else to provide here.

```
Building: board/gen/mega2560/Pins.cpp
cc1plus: internal compiler error: Illegal instruction: 4
libbacktrace could not find executable to open
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://github.com/osx-cross/homebrew-avr/issues> for instructions.
make[1]: *** [build/app/mega2560/release/board/gen/mega2560/Pins.cpp.o] Error 1
```

```
avr-gcc (Homebrew AVR GCC 9.3.0_2) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

ladislas, 2021-02-10:
What's the output of `brew doctor`

Can you try `brew remove avr-gcc avr-binutils && brew upgrade && brew install avr-gcc`?

paradajz, 2021-02-10:
doctor output:
```

paradajzs-MacBook-Pro:src paradajz$ brew doctor
Please note that these warnings are just used to help the Homebrew maintainers
with debugging if you file an issue. If everything you use Homebrew for is
working fine: please don't worry or file an issue; just ignore this. Thanks!

Warning: Homebrew's "sbin" was not found in your PATH but you have installed
formulae that put executables in /usr/local/sbin.
Consider setting your PATH for example like so:
  echo 'export PATH="/usr/local/sbin:$PATH"' >> ~/.profile

Warning: Broken symlinks were found. Remove them with `brew cleanup`:
  /usr/local/bin/code
```

Nothing changed after running what you suggested.

ladislas, 2021-02-10:
Was it working before? Can you try on another project?

paradajz, 2021-02-10:
Well, I'm running Catalina on MacBook Pro 2010 (installed through dosdude1 patch, if it matters - it was running clean install of High Sierra fist, then I upgraded it to Catalina). It worked few months ago before the upgrade - I'm not really using this laptop actively. The project is hosted here: https://github.com/paradajz/OpenDeck

How can i build the toolchain from source?

ladislas, 2021-02-10:
> How can i build the toolchain from source?

`brew remove avr-gcc && brew install avr-binutils --head && brew install avr-gcc --head`

paradajz, 2021-02-10:
I'm getting:

    Error: No head is defined for osx-cross/avr/avr-binutils
    Error: No head is defined for osx-cross/avr/avr-gcc@9

ladislas, 2021-02-10:
ah right... I removed that...

Do you have another mac where you could test? A little more recent?

ladislas, 2021-02-10:
I've just tested on my machine, it works great, I don't have the same issue.

Can you try another version? @8 or @10 to see if it's related?

paradajz, 2021-02-10:
Both 8 and 10 act the same. I'm going to assume that is is something related to unofficial update to Catalina.

ladislas, 2021-02-10:
Is your Xcode up to date?

paradajz, 2021-02-10:
This could be of relevance:
https://github.com/conda/conda/issues/9678

I assume the binaries you provide are compiled on newer machines which probably have newer CPU features, which my Mac doesn't.

ladislas, 2021-02-10:
0ed0db6 adds back the ability to build with from source with `--head`.

Can you try again and let me know? Compilation can take between 30min and 3~4 hours depending on your machine. You can use the `-v` to see the output and make sure it's not stuck.

paradajz, 2021-02-10:
I've tried with `brew install avr-gcc --build-from-source` flag an hour ago and it failed with same Illegal instruction 4 error after a while. I took it as a definite answer to "will Catalina work fine on this old MacBook" (it won't). I've reverted back to 10.13 High Sierra and running the install right now, will let you know. Since most of the things are now built from source (10.13 isn't officially supported by Homebrew anymore) it will take a while.

paradajz, 2021-02-11:
Only 16 hours later I have everything setup. :) Had to build nearly everything I need from source since almost no bottles exist anymore for High Sierra. The toolchain works, I can compile my project. Sorry for opening non-issue issue.

closed 2021-02-11 by paradajz

ladislas, 2021-02-11:
no worries! glad you fixed it :)