osx-cross/homebrew-avr #230: There is no `/usr/bin/sed` on GNU/Linux

liyang, 2021-03-06:
fcd39d44 adds 
```
    # Avoid reference to sed shim
    args << "SED=/usr/bin/sed"
```
to the `configure` args, but unfortunately on GNU/Linux `sed` lives at `/bin/sed` (Ubuntu 16.04 here, so at least all Debian-based distros). I'm not sure where ‘sed shim’ the comment refers to lives on macOS, so unfortunately I can't realistically provide a patch.

Commenting out the above line builds for me.

ladislas, 2021-03-06:
Thanks @liyang for reporting this issue. I don't use Linux and the tap doesn't officially support it.

But the fix seems simple enough with a check of the current OS and add or not the arg.

Could you provide a PR?

liyang, 2021-03-06:
> Thanks @liyang for reporting this issue. I don't use Linux and the tap doesn't officially support it.

You seem to be my only source for a semi-effortless install of recent AVR GCC for Ubuntu/Debian. :) I couldn't find a PPA anywhere! Other than this issue, it works flawlessly.

> But the fix seems simple enough with a check of the current OS and add or not the arg.
> Could you provide a PR?

I can try, but I don't have macOS here (except a dusty 2010 MacBook Air), so I can't guarantee it won't break anything!

Do you remember why you added that override, or what the `sed shim` your comment refers to is? i.e. did you want to avoid the homebrew copy of `sed` and use the one that ships with macOS? And if so, what problem was the homebrew `sed` causing?

liyang, 2021-03-06:
> I don't have macOS here

Nevermind. I see there's a CI `brew test-bot`. :)

liyang, 2021-03-06:
So I enabled `--verbose` in the `brew test-bot` script, added `depends_on "gnu-sed"` & removed the `SED=/usr/bin/sed` override, and I'm seeing `configure` pick up
```
checking for a sed that does not truncate output... /usr/local/opt/gnu-sed/bin/gsed
```
https://github.com/osx-cross/homebrew-avr/runs/2047242809#step:8:571

so GCC builds just fine on macOS.

A [previous run _without_ `gnu-sed`](https://github.com/osx-cross/homebrew-avr/runs/2047089830) (no `--verbose` unfortunately) failed with:
```
==> brew audit osx-cross/avr/avr-gcc@8 --online --git --skip-style
==> FAILED
osx-cross/avr/avr-gcc@8:
  * Files were found with references to the Homebrew shims directory.
    The offending files are:
      libexec/gcc/avr/8.4.0/install-tools/fixincl
Error: Files were found with references to the Homebrew shims directory.
The offending files are:
  libexec/gcc/avr/8.4.0/install-tools/fixincl

Error: 1 problem in 1 formula detected

==> brew bottle --verbose --json osx-cross/avr/avr-gcc@8 --root-url=https://github.com/osx-cross/homebrew-avr/releases/download/avr-gcc@8-8.4.0_1
==> FAILED
==> Determining osx-cross/avr/avr-gcc@8 bottle rebuild...
==> Bottling avr-gcc@8--8.4.0_1.catalina.bottle.1.tar.gz...
/usr/bin/sudo --non-interactive /usr/sbin/purge
tar cf /usr/local/Homebrew/Library/Taps/osx-cross/homebrew-avr/avr-gcc@8--8.4.0_1.catalina.bottle.1.tar avr-gcc@8/8.4.0_1
/usr/bin/sudo --non-interactive /usr/sbin/purge
gzip -f avr-gcc@8-bottle.tar
/usr/bin/sudo --non-interactive /usr/sbin/purge
==> Detecting if avr-gcc@8--8.4.0_1.catalina.bottle.1.tar.gz is relocatable...
Warning: String '/usr/local/Homebrew' still exists in these files:
/usr/local/Cellar/avr-gcc@8/8.4.0_1/libexec/gcc/avr/8.4.0/install-tools/fixincl
 --> match '/usr/local/Homebrew/Library/Homebrew/shims/mac/super/sed' at offset 0x10720
Error: Bottle contains non-relocatable reference to /usr/local/Homebrew!
```
So it seems that _without_ `gnu-sed` installed, `configure` disregards the macOS-shipped `/usr/bin/sed` and uses [this shim that works around an issue with a barfing lion on some mountain](https://github.com/Homebrew/brew/blob/master/Library/Homebrew/shims/mac/super/sed), and compiles the path of _that_ into [install-tools/fixincl](https://gcc.gnu.org/git/?p=gcc.git;a=blob;f=fixincludes/fixincl.tpl;h=3d70cabc7fd6dd0e63033e6523fac1dca01b285a;hb=HEAD#l42), which is arguably the correct thing to do: otherwise Mountain Lion users would barf if `fixincl` had to do its thing on any header files that included Unicode characters. (UTF-8? UTF-16? UCS-4? Dunnolol. [This is where the shim was added](https://github.com/Homebrew/brew/commit/4f06d131dfffc9de625697c2135262f624534a6e), and [this is as far back as I could trace that workaround](https://github.com/Homebrew/brew/commit/65d195dcaaa4304e127e0a264cf97ad7b6a7fd83#diff-9bc613a2613e2d1d4f9d20e6f8f8833d3effbe7e159cda0c80ca464bfaad49afR41)).

I suggest we just `depends_on "gnu-sed" if OS.mac?` and avoid the above issue entirely. Will send the PR shortly.

ladislas, 2021-03-06:
Thanks for investigating, I'll reply on the PR.

ladislas, 2021-10-05:
@liyang I'll close the issue for now, feel free to reopen if needed.

closed 2021-10-05 by ladislas