osx-cross/homebrew-avr #232: arm64 support / M1

braindaddy73, 2021-05-03:
Hi there,

is it possible to build the toolchain for the new M1?

Thx
Heiko

ladislas, 2021-05-03:
Unfortunately not yet. We do not have access to any M1 Macs. But it should work as is with Rosetta 2

braindaddy73, 2021-05-04:
Okay I will try. It didn't install because of required architecture. But I am new to homebrew and perhaps I find a way. :P Perhaps by time i will try to compile from source. To much work at the moment ...

ladislas, 2021-05-04:
Try the simplest version first. It should work without the need to recompile from source.

braindaddy73, 2021-05-04:
Ok, works like a charm. Had to install the homebrew-prefix for x86_64 first. Then point to the right brew installation:

arch -x86_64 /bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install.sh)"
arch -x86_64 /usr/local/bin/brew tap osx-cross/avr
arch -x86_64 /usr/local/bin/brew install avr-gcc
arch -x86_64 /usr/local/bin/brew install avr-gdb

Thx for help!

closed 2021-05-04 by ladislas