osx-cross/homebrew-avr #233: Support for avr-gcc 11

bartekpacia, 2021-05-25:
As the title says. Will (if yes, then when?) `avr-gcc@11` be available?

ladislas, 2021-05-25:
GCC 11 is supposed to remove support for AVR, so I don't know. I'm open to a PR if you want to try.

https://www.phoronix.com/scan.php?page=news_item&px=GCC-11-Dropping-CC0

bartekpacia, 2021-05-26:
Hmmm, that link you gave is interesting, even more given the fact that yesterday I installed `avr-gcc` on my Windows machine from [this website](https://blog.zakkemble.net/avr-gcc-builds/). It is v11.1.0 and works just fine.

ladislas, 2021-05-27:
That's great news then. I don't have much time currently to add the formula but I can review a PR. You can look at the commit history for examples.

Also look at how they do it here: https://github.com/Homebrew/homebrew-core/blob/master/Formula/gcc.rb

closed 2021-05-29 by bartekpacia