osx-cross/homebrew-avr #236: Missing support for newer MCUs

bartekpacia, 2021-05-29:
I work on one AVR project that's targeting [AVR128DB28](https://www.microchip.com/wwwproducts/en/AVR128DB28) and have found out that `avr-gcc` (even the newest version 11) doesn't support it (i.e `avr/io.h` is not defined for it in `/usr/local/Cellar/avr-gcc@11/11.1.0_1/avr/include/avr`).

Here's a [related avr freaks thread](https://www.avrfreaks.net/forum/avr-gcc-not-being-updated-new-devices#comment-3146646).

Is there any chance that header files for those newer MCUs will be added to `avr-libc`? I know this is not necessarily the place to ask this question but as of now, I'm just not sure where else to ask this.

ladislas, 2021-05-30:
Contrary to this [comment](https://www.avrfreaks.net/comment/3146851#comment-3146851), it's not up to "us" to package anything. This formula is just an easy way to install stuff developed by others.

The only real solution is to provide a patch for avr-libc and hope it will be merged. Now, there isn't much activity on avr-libc, so it may take quite some time.

So we need a workaround. One way to do it is just like in https://github.com/osx-cross/homebrew-avr/pull/105. You can use the PR as a reference to provide the needed patch as an option. You won't be able to have a bottle but it's still better than nothing.

Another way could be to fork avr-libc and merge the patch there, see if the community wants to take over.

bartekpacia, 2021-05-30:
Thank you. It looks like that `avr-libc` on https://savannah.nongnu.org is pretty much dead. I'll try to copy the solution from #105

Feuerwerk, 2021-06-10:
Since i recently had a look at my old notes und currently working on fixing #245 i had a quick look at the Atmel Packs to find your chip. At the weekend i will give it a try to add also add support for this one.

@ladislas If i'm successful, should i introduce a second option (e.g. --with-AVR128DB28Support) or should i add a more generic option (e.g. --with-UnsupportedMCUs) and apply both patches at the same time (and maybe further patches in the  future)

Feuerwerk, 2021-06-10:
For documentation, these are the steps i did when creating the patch for #105 

Patching avr-libc

- add include/avr/iom168pb.h from Atmel Atmega Packs (Atmel.ATmega_DFP.1.3.300/include/avr)
- devtools/gen-avr-lib-tree.sh
    - Line 199
        - atmega168pb:crtm168pb.o:${DEV_DEFS}:${CFLAGS_SPACE}:${DEV_ASFLAGS};\
- avr/lib/avr5/Makefile.am
    - Add atmega168pb to SUBDIRS in Line 31
    - Line 143:
        - if HAS_atmega168pb
        -   AVRLIB_DEVLIST += atmega168pb
        - endif	# atmega168pb
- configure.ac
    - Line 823:
        - CHECK_AVR_DEVICE(atmega168pb)
        - AM_CONDITIONAL(HAS_atmega168pb, test "x$HAS_atmega168pb“ = "xyes")
    - Line 1465:
        - avr/lib/avr5/atmega168pb/Makefile
- include/avr/io.h
    - Line 299
        - #elif defined (__AVR_ATmega168PB__)
        - #  include <avr/iom168pb.h>
- include/avr/Makefile.am
    - Line 114
        - iom168pb.h \

- add findreg.xsl and ioreg.pl from Source Repository to devtools folder
- ./bootstrap
- cd devtools
- ./ioreg.pl -o ../crt1/iosym/atmega168pb.S  ~/Downloads/Atmel.ATmega_DFP.1.3.300/atdf/ATmega168PB.atdf
- cd ..
- ./configure --build=i686-apple-darwin18.2.0 --prefix=/usr/local/gurke --host=avr
- make -j 4
- sudo make install
- 
patch -p1 < ../avr-libc-2.0.0_atmega168pb.patch

ladislas, 2021-06-10:
> If i'm successful, should i introduce a second option (e.g. --with-AVR128DB28Support) or should i add a more generic option (e.g. --with-UnsupportedMCUs) and apply both patches at the same time (and maybe further patches in the future)

@Feuerwerk one option sounds good! I'll go with `--with-newer-mcus`

It would be also good to add a test workflow for the option to make sure it works for future releases. :)

ladislas, 2021-10-05:
I'll close this issue for now, feel free to open PR to add new MCUs if needed, I'll happily review them.

closed 2021-10-05 by ladislas