osx-cross/homebrew-avr #237: avr-gcc@9: The x86_64 architecture is required for this software.

brucehoult, 2021-06-04:
As above.

$ brew install avr-gcc avr-binutils avrdude

... fails on ARM Macs.

ladislas, 2021-06-04:
brew install avr-gcc avrdude should be sufficient (binutils is pulled automatically)

have you installed Rosetta 2 on your mac?

brucehoult, 2021-06-04:
Yes, but it seems to be a while lot of trouble to run both arm and x86 homebrew together.

Mac-mini:programs bruce$ arch -x86_64 brew install avr-gcc
Error: Cannot install under Rosetta 2 in ARM default prefix (/opt/homebrew)!
To rerun under ARM use:
    arch -arm64 brew install ...
To install under x86_64, install Homebrew into /usr/local.

Is it not possible to get an ARM version? AVR toolchain works fine (but slowly) on a Raspberry Pi.

ladislas, 2021-06-04:
No it's not possible as I don't have an Apple silicon Mac, nor does the CI. 

You can build from source with the --HEAD option. 

You might need avr-gcc 11 in this case though.

ladislas, 2021-06-04:
I have bumped all the formula, maybe it can help.

DavidEGrayson, 2021-06-11:
I have a Mac Mini with the M1 chip and I am also having this error.  I'm not sure what command to run to install avr-gcc, but here is a shell session showing some that I tried:

```text
% brew tap osx-cross/avr 
% brew install avr-gcc   
==> Installing avr-gcc@9 from osx-cross/avr
avr-gcc@9: The x86_64 architecture is required for this software.
Error: An unsatisfied requirement failed this build.
% brew install --head avr-gcc
==> Installing avr-gcc@9 from osx-cross/avr
avr-gcc@9: The x86_64 architecture is required for this software.
Error: An unsatisfied requirement failed this build.
% brew install --head avr-gcc@10
==> Installing avr-gcc@10 from osx-cross/avr
avr-gcc@10: The x86_64 architecture is required for this software.
Error: An unsatisfied requirement failed this build.
% brew install --head avr-gcc@11
==> Installing avr-gcc@11 from osx-cross/avr
avr-gcc@11: The x86_64 architecture is required for this software.
Error: An unsatisfied requirement failed this build.
```

Does anyone have other suggestions of what to try?

ladislas, 2021-06-11:
The problem is with the bottle and the place where it must be installed by brew to work correctly and find all the tools it needs. 

You can either have two brew, one for Rosetta and one for the M1 as explained here: https://stackoverflow.com/questions/64882584/how-to-run-the-homebrew-installer-under-rosetta-2-on-m1-macbook

Or you can simply build from source with the -s option. 
It should be quite fast on the M1.

ladislas, 2021-06-11:
Well I think I found the solution! 

Can you guys try again with removing the `depends_on arch: :x86_64` line in the formula you want to install? 

It should work and if it does I'll make a PR with the changes.

DavidEGrayson, 2021-06-11:
Thank you for the idea.  If I clone this repository, edit the formulae to remove the line that references `x86_64`, and then try to build GCC 9, 10, or 11, I get an error because it cannot find a symbol named `_host_hooks`.  I found a [discussion about this error](https://github.com/riscv/riscv-gnu-toolchain/issues/800) where people are saying that GCC does not yet support running on an ARM Darwin machines.

 I tried adding `--disable-pch` and `--disable-lto` to the GCC configure flags but that was not sufficient to fix the problem.

```text
% brew install --formula ./Formula/avr-gcc@11.rb
==> Downloading https://download.savannah.gnu.org/releases/avr-libc/avr-libc-2.0.0.tar.bz2
Already downloaded: /Users/david/Library/Caches/Homebrew/downloads/02c166efff9cac7ac9108da0a404d5ab73732ab0d785909ed7bc76e2e6b953da--avr-libc-2.0.0.tar.bz2
==> Downloading https://ftp.gnu.org/gnu/gcc/gcc-11.1.0/gcc-11.1.0.tar.xz
Already downloaded: /Users/david/Library/Caches/Homebrew/downloads/104cdaaeb30152af30fd8af2d3da5eb879ba1191832e8225d91f3538e4002fb6--gcc-11.1.0.tar.xz
Warning: A newer Command Line Tools release is available.
Update them from Software Update in System Preferences or run:
  softwareupdate --all --install --force

If that doesn't show you any updates, run:
  sudo rm -rf /Library/Developer/CommandLineTools
  sudo xcode-select --install

Alternatively, manually download them from:
  https://developer.apple.com/download/more/.
You should download the Command Line Tools for Xcode 12.5.

==> ../configure --target=avr --prefix=/opt/homebrew/Cellar/avr-gcc@11/11.1.0_1 --libdir=/opt/homebrew/C
==> make BOOT_LDFLAGS=-Wl,-headerpad_max_install_names
Last 15 lines from /Users/david/Library/Logs/Homebrew/avr-gcc@11/02.make:
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lto1] Error 1
make[2]: *** Waiting for unfinished jobs....
Undefined symbols for architecture arm64:
  "_host_hooks", referenced from:
      gt_pch_save(__sFILE*) in libbackend.a(ggc-common.o)
      gt_pch_restore(__sFILE*) in libbackend.a(ggc-common.o)
      toplev::main(int, char**) in libbackend.a(toplev.o)
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lto-dump] Error 1
rm gcc.pod
make[1]: *** [all-gcc] Error 2
make: *** [all] Error 2

Do not report this issue to Homebrew/brew or Homebrew/core!

Traceback (most recent call last):
	30: from /opt/homebrew/Library/Homebrew/build.rb:229:in `<main>'
	29: from /opt/homebrew/Library/Homebrew/build.rb:133:in `install'
	28: from /opt/homebrew/Library/Homebrew/utils.rb:558:in `with_env'
	27: from /opt/homebrew/Library/Homebrew/build.rb:138:in `block in install'
	26: from /opt/homebrew/Library/Homebrew/formula.rb:1259:in `brew'
	25: from /opt/homebrew/Library/Homebrew/formula.rb:2344:in `stage'
	24: from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/forwardable.rb:230:in `stage'
	23: from /opt/homebrew/Library/Homebrew/resource.rb:91:in `stage'
	22: from /opt/homebrew/Library/Homebrew/resource.rb:116:in `unpack'
	21: from /opt/homebrew/Library/Homebrew/resource.rb:199:in `mktemp'
	20: from /opt/homebrew/Library/Homebrew/mktemp.rb:63:in `run'
	19: from /opt/homebrew/Library/Homebrew/mktemp.rb:63:in `chdir'
	18: from /opt/homebrew/Library/Homebrew/mktemp.rb:63:in `block in run'
	17: from /opt/homebrew/Library/Homebrew/resource.rb:117:in `block in unpack'
	16: from /opt/homebrew/Library/Homebrew/download_strategy.rb:102:in `stage'
	15: from /opt/homebrew/Library/Homebrew/download_strategy.rb:115:in `chdir'
	14: from /opt/homebrew/Library/Homebrew/download_strategy.rb:115:in `chdir'
	13: from /opt/homebrew/Library/Homebrew/resource.rb:121:in `block (2 levels) in unpack'
	12: from /opt/homebrew/Library/Homebrew/formula.rb:2364:in `block in stage'
	11: from /opt/homebrew/Library/Homebrew/utils.rb:558:in `with_env'
	10: from /opt/homebrew/Library/Homebrew/formula.rb:2365:in `block (2 levels) in stage'
	 9: from /opt/homebrew/Library/Homebrew/formula.rb:1266:in `block in brew'
	 8: from /opt/homebrew/Library/Homebrew/build.rb:178:in `block (2 levels) in install'
	 7: from /Users/david/homebrew-avr/Formula/avr-gcc@11.rb:109:in `install'
	 6: from /opt/homebrew/Library/Homebrew/formula.rb:2259:in `mkdir'
	 5: from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/fileutils.rb:128:in `cd'
	 4: from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/fileutils.rb:128:in `chdir'
	 3: from /Users/david/homebrew-avr/Formula/avr-gcc@11.rb:115:in `block in install'
	 2: from /opt/homebrew/Library/Homebrew/formula.rb:2135:in `system'
	 1: from /opt/homebrew/Library/Homebrew/formula.rb:2135:in `open'
/opt/homebrew/Library/Homebrew/formula.rb:2199:in `block in system': Failed executing: make BOOT_LDFLAGS=-Wl,-headerpad_max_install_names (BuildError)
	5: from /opt/homebrew/Library/Homebrew/brew.rb:155:in `<main>'
	4: from /opt/homebrew/Library/Homebrew/brew.rb:167:in `rescue in <main>'
	3: from /opt/homebrew/Library/Homebrew/exceptions.rb:494:in `dump'
	2: from /opt/homebrew/Library/Homebrew/exceptions.rb:440:in `issues'
	1: from /opt/homebrew/Library/Homebrew/exceptions.rb:444:in `fetch_issues'
/opt/homebrew/Library/Homebrew/utils/github.rb:66:in `issues_for_formula': undefined method `full_name' for nil:NilClass (NoMethodError)
```

ladislas, 2021-06-11:
Then I can't really help. Take a look at gcc.rb in brew/core, they build for Apple silicon so maybe the answer is there.

ladislas, 2021-06-23:
This was fixed by #248 / #249 and fd5d73f (thanks @DavidEGrayson)

Hope this helps!

closed 2021-06-23 by ladislas