osx-cross/homebrew-avr #243: avr-binutils conlifct with existing x86_64-elf-binutils

gustavonihei, 2021-06-04:
After the bump of avr-binutils to version 2.36.1, it seems to conflict with x86_64-elf-binutils installation of the same 2.36.1 version:

```
==> Pouring avr-binutils-2.36.1.catalina.bottle.tar.gz
Error: The `brew link` step did not complete successfully
The formula built, but is not symlinked into /usr/local
Could not symlink lib/bfd-plugins/libdep.so
Target /usr/local/lib/bfd-plugins/libdep.so
is a symlink belonging to x86_64-elf-binutils. You can unlink it:
  brew unlink x86_64-elf-binutils

To force the link and overwrite all conflicting files:
  brew link --overwrite avr-binutils

To list all files that would be deleted:
  brew link --overwrite --dry-run avr-binutils

Possible conflicting files are:
/usr/local/lib/bfd-plugins/libdep.so -> /usr/local/Cellar/x86_64-elf-binutils/2.36.1/lib/bfd-plugins/libdep.so
```

ladislas, 2021-06-05:
This is really strange, I don't know why they would conflict.

ladislas, 2021-06-05:
I've tried something, but I'm really not sure on how to fix this.

Can you try fcc36c4 and let me know?

ralphrmartin, 2021-06-06:
Similar happens with 
arm-linux-gnueabihf-binutils

See https://github.com/Homebrew/homebrew-core/issues/78849

gustavonihei, 2021-06-07:
> I've tried something, but I'm really not sure on how to fix this.
> 
> Can you try [fcc36c4](https://github.com/osx-cross/homebrew-avr/commit/fcc36c4e399d145a7231670679e674d79055b326) and let me know?

I've just tested it, but it didn't resolve the conflict.

One important detail that I've realized is that the `libdep.so` symlink is not created during the install of `x86_64-elf-gcc`, but during the install of `avr-gcc`. Even so, the symlink points to the `libdep.so` of `x86_64-elf-binutils` package.

Here is the steps for reproducing the issue:
```bash
$ brew uninstall avr-gcc avr-binutils
Uninstalling /usr/local/Cellar/avr-gcc@9/9.3.0_3... (1,749 files, 207.3MB)
Uninstalling /usr/local/Cellar/avr-binutils/2.36.1... (159 files, 23.4MB)

$ brew uninstall x86_64-elf-gcc x86_64-elf-binutils
Uninstalling /usr/local/Cellar/x86_64-elf-gcc/11.1.0... (615 files, 129.8MB)
Uninstalling /usr/local/Cellar/x86_64-elf-binutils/2.36.1... (189 files, 34.6MB)

$ brew install x86_64-elf-gcc
==> Downloading https://ghcr.io/v2/homebrew/core/x86_64-elf-binutils/manifests/2.36.1
==> Downloading https://ghcr.io/v2/homebrew/core/x86_64-elf-binutils/blobs/sha256:98a1c41b67cda5c8552edf16e826aada8f29409da4477301db7ce861c6e93295
==> Downloading https://ghcr.io/v2/homebrew/core/x86_64-elf-gcc/manifests/11.1.0-1
==> Downloading https://ghcr.io/v2/homebrew/core/x86_64-elf-gcc/blobs/sha256:c5e485f635596a59826e07708ccc497cc950503a4711fe83faf9aafa5ebeb914
==> Installing dependencies for x86_64-elf-gcc: x86_64-elf-binutils
==> Installing x86_64-elf-gcc dependency: x86_64-elf-binutils
==> Pouring x86_64-elf-binutils--2.36.1.big_sur.bottle.tar.gz
🍺  /usr/local/Cellar/x86_64-elf-binutils/2.36.1: 189 files, 34.6MB
==> Installing x86_64-elf-gcc
==> Pouring x86_64-elf-gcc--11.1.0.big_sur.bottle.1.tar.gz
🍺  /usr/local/Cellar/x86_64-elf-gcc/11.1.0: 615 files, 129.8MB

$ find /usr/local/ -name libdep.so
/usr/local//Cellar/x86_64-elf-binutils/2.36.1/lib/bfd-plugins/libdep.so

$ cd $(brew --repository)/Library/Taps/osx-cross/homebrew-avr

$ git checkout fcc36c4
HEAD is now at fcc36c4 Set --libdir - fix #243

$ HOMEBREW_NO_AUTO_UPDATE=1 brew install avr-gcc
==> Installing avr-gcc@9 from osx-cross/avr
==> Downloading https://github.com/osx-cross/homebrew-avr/releases/download/avr-binutils-2.36.1/avr-binutils-2.36.1.big_sur.bottle.tar.gz
==> Downloading from https://github-releases.githubusercontent.com/30939945/d4e01980-c53f-11eb-8ce4-639a0401e406?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSV
######################################################################## 100.0%
==> Downloading https://github.com/osx-cross/homebrew-avr/releases/download/avr-gcc@9-9.3.0_3/avr-gcc%409-9.3.0_3.big_sur.bottle.tar.gz
==> Downloading from https://github-releases.githubusercontent.com/30939945/f514d600-c549-11eb-8afb-57a7015a2f3f?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSV
######################################################################## 100.0%
==> Installing dependencies for osx-cross/avr/avr-gcc@9: avr-binutils
==> Installing osx-cross/avr/avr-gcc@9 dependency: avr-binutils
==> Pouring avr-binutils-2.36.1.big_sur.bottle.tar.gz
Error: The `brew link` step did not complete successfully
The formula built, but is not symlinked into /usr/local
Could not symlink lib/bfd-plugins/libdep.so
Target /usr/local/lib/bfd-plugins/libdep.so
is a symlink belonging to x86_64-elf-binutils. You can unlink it:
  brew unlink x86_64-elf-binutils

To force the link and overwrite all conflicting files:
  brew link --overwrite avr-binutils

To list all files that would be deleted:
  brew link --overwrite --dry-run avr-binutils

Possible conflicting files are:
/usr/local/lib/bfd-plugins/libdep.so -> /usr/local/Cellar/x86_64-elf-binutils/2.36.1/lib/bfd-plugins/libdep.so
==> Summary
🍺  /usr/local/Cellar/avr-binutils/2.36.1: 159 files, 23.4MB
==> Installing osx-cross/avr/avr-gcc@9
==> Pouring avr-gcc
==> Caveats
For Mac computers with Apple silicon, avr-gcc might need Rosetta 2 to work properly.
You can learn more about Rosetta 2 here:
    > https://support.apple.com/en-us/HT211861
==> Summary
🍺  /usr/local/Cellar/avr-gcc@9/9.3.0_3: 1,749 files, 207.3MB
==> Caveats
==> avr-gcc@9
For Mac computers with Apple silicon, avr-gcc might need Rosetta 2 to work properly.
You can learn more about Rosetta 2 here:
    > https://support.apple.com/en-us/HT211861

$ find /usr/local/ -name libdep.so
/usr/local//lib/bfd-plugins/libdep.so
/usr/local//Cellar/x86_64-elf-binutils/2.36.1/lib/bfd-plugins/libdep.so
/usr/local//Cellar/avr-binutils/2.36.1/lib/bfd-plugins/libdep.so
```

ladislas, 2021-06-07:
@gustavonihei You must install binutils from source. The commit I sent does not point to a bottle yet. Use the -s option

ladislas, 2021-06-07:
@ralphrmartin Thanks for the info. The update they mention for this formula was updating binutils from 2.35 to 2.36. Maybe this caused the issue but I don't know how. 

Do elf and gnueabielf conflit with one another? Can you try?

ladislas, 2021-06-07:
@gustavonihei I've forced push to the branch to remove the bottle block and force building from source, commit is now f1cbdc2

ralphrmartin, 2021-06-07:
"Do elf and gnueabielf conflict with one another? Can you try?"

I basically have a choice of leaving avr-binutils as it is, or overwriting with arm-linux-gnueabihf-binutils. 
I am not quite sure how to test for a conflict, but if you give me some specific instructions, I'll certainly try to find out.

ladislas, 2021-06-07:
I'm ooo at the moment, so I cannot really try myself. The idea is:

```bash
# remove everythting completely
brew uninstall --ignore-dependencies avr-binutils
brew uninstall --ignore-dependencies x86_64-elf-binutils
brew uninstall --ignore-dependencies arm-linux-gnueabihf-binutils

# install x86_64-elf-binutils --> it should not complain
brew install x86_64-elf-binutils

# install arm-linux-gnueabihf-binutils
brew install arm-linux-gnueabihf-binutils
```

If there is a conflict, then it should be fixed in brew/core as both formula are located there. I'll follow the same fix here when they provide one.

ralphrmartin, 2021-06-07:
Yes, I tested it too and got the conflict

gustavonihei, 2021-06-07:
> @gustavonihei I've forced push to the branch to remove the bottle block and force building from source, commit is now [f1cbdc2](https://github.com/osx-cross/homebrew-avr/commit/f1cbdc285df410afab96381298ab09a63e930e16)

Thanks, I've tested at this revision and `avr-binutils` installed successfully alongside `x86_64-elf-binutils`. I didn't investigate to understand what is actually done in the install process, but now the symlink at `/usr/local/lib/bfd-plugins/libdep.so` has not been created:

```shell
$ find /usr/local/ -name libdep.so
/usr/local//Cellar/x86_64-elf-binutils/2.36.1/lib/bfd-plugins/libdep.so
/usr/local//Cellar/avr-binutils/2.36.1/avr-binutils/bfd-plugins/libdep.so

$ x86_64-elf-ld
x86_64-elf-ld: no input files

$ avr-ld
avr-ld: no input files
```

commit f1cbdc28 (2021-06-07): Set --libdir - fix #243
commit 89bf6c54 (2021-06-08): Set --libdir - fix #243

Closes #244.
closed 2021-06-08 by ladislas