osx-cross/homebrew-avr #245: URL for ATMega168pbSupport-Patch is forbidden

Feuerwerk, 2021-06-09:
Hi,

Installing avr-gcc with --with-ATMega168pbSupport activated one will get a 403 forbidden error when home-brew try to download the patch https://dl.bintray.com/osx-cross/avr-patches/avr-libc-2.0.0-atmega168pb.patch

ladislas, 2021-06-09:
Yes, bintray has closed its doors...
We should move the patch here, can you make a PR

https://github.com/osx-cross/homebrew-avr/tree/master/Patch

ladislas, 2021-06-09:
ideally, we would add the patch in a first PR, then reference the patch raw link in another PR.

Feuerwerk, 2021-06-09:
Can't we reference https://github.com/osx-cross/homebrew-avr/blob/master/Patch/avr-binutils-size.patch directly?
Whats the benefit to reference the patch inside a PR?

Just for clarification how this is working:
- I create a PR which only includes the patch
- This PR is rejected
- Then i create a second PR updating the brew-receipt referencing to the patch inside the first PR

Is that correct?

ladislas, 2021-06-09:
> Can't we reference https://github.com/osx-cross/homebrew-avr/blob/master/Patch/avr-binutils-size.patch directly?
> 
> Whats the benefit to reference the patch inside a PR?

I don't think it's the same patch. The one already here is just for size in binutils but it actually not used. 

> Just for clarification how this is working:
> 
> - I create a PR which only includes the patch
> 
> - This PR is rejected
> 
> - Then i create a second PR updating the brew-receipt referencing to the patch inside the first PR
>
> Is that correct?

The first PR will be accepted and merged. Then you can use the reference from master for the second PR

Feuerwerk, 2021-06-09:
Ahhh, ok, got it. I think i still have the original (small) patch, i need to have a look and than create the PR

PR #246 (2021-06-10): Add avr-libc patch for ATMega168pb mcu support
First part to fix #245
closed 2021-06-10 by ladislas