osx-cross/homebrew-avr #251: avr-gdb: Symbol format `elf32-avr' unknown.

mmastrac, 2021-06-24:
I've been scratching my head over this for a while, but I cannot get the `avr-gdb` package here to load an elf file successfully. I tried a known-good incantation from http://blog.oddbit.com/post/2019-01-22-debugging-attiny-code-pt-1/, but something about elf32-avr is busted.

This is a repro with a simple, empty main.c (int main(void) { }):

```
$ avr-gcc -mmcu=atmega328p /tmp/main.c -o /tmp/main
$ avr-gdb /tmp/main
GNU gdb (GDB) 10.1
[snip]

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /tmp/main...
I'm sorry, Dave, I can't do that.  Symbol format `elf32-avr' unknown.
```

ladislas, 2021-06-24:
I haven't used gdb in a while so I can really tell. 

Maybe you're right this line causes the error. You can comment it locally and build from source to test it. If it works I'll make a fix.

mmastrac, 2021-06-24:
Unfortunately, I tried a few different edits after posting that message (removing various combinations of --disable-*, install vs install-gdb) and had no luck. 

It's busted, but I am really at a loss as to how it might be fixed. Hopefully someone with more experience can point me at some ideas!

felias-fogg, 2021-07-13:
I also had this problem and found this clue: https://unix.stackexchange.com/questions/624486/gdb-installed-from-homebrew-on-macos-big-sur-fails-to-load-symbols-from-elf32-i3

So, I tried to insert the line "#include <string.h>" at the top of bfd/elf-bfd.h and it did the job. So, in order to fix this bug, you need to include the following line in the homebrew formula right after 'mkdir "build" do':
         inreplace "../bfd/elf-bfd.h", "#define _LIBELF_H_ 1", "#define _LIBELF_H_ 1\n#include <string.h>"
And when you are at it, you may also want to use the new gdb version 10.2.

Cheers,
Bernhard

danapplegate, 2021-07-30:
I'm encountering this same issue, and unfortunately @felias-fogg 's solution hasn't worked for me. I'm on MacOS Big Sur version 11.2.3.

Neither adding the `inreplace` line into the formula, nor including the `patch` block from #255 worked for me.

danapplegate, 2021-07-30:
Actually, scratch that, I was being dumb. The patch inserts from #255 worked for me, I just needed to `--build-from-source` when installing `avr-binutils` and `avr-gdb`. Now seems to work for me!

akasakakona, 2021-09-29:
I encountered this error when recompiling, can anyone help?
```
Error: An exception occurred within a child process:
  Utils::Inreplace::Error: inreplace failed
../bfd/elf-bfd.h:
  expected replacement of "#define LIBELF_H 1" with "#define LIBELF_H 1\n#include <string.h>"
  ```

ladislas, 2021-10-05:
Does someone want to take over #255 and the patch?

danapplegate, 2021-10-05:
@ladislas Sure thing, here you go: #259

ladislas, 2021-10-07:
This should now be fixed thanks to @danapplegate! 

I'll close the issue now.

Feel free to open a new one if needed.