osx-cross/homebrew-avr #257: Linking with stl

SpyrosMourelatos, 2021-09-09:
Thank  you for the amazing job , you saved me some hours but I have to ask( it's more of a question and less of an issue):
How we link to `<atomic>`s ? I can't find an implementation of stl for avr's on the internet and I hoped you had some insight
Thanks again

ladislas, 2021-09-09:
it's a system header, so if it's available your code will compile, and if not it will tell you so.

`std::atomic` is available with `-std=c++11`

Here are some interesting reads:

- https://www.nongnu.org/avr-libc/user-manual/group__util__atomic.html
- https://www.avrfreaks.net/forum/atomic-oo-way-0

SpyrosMourelatos, 2021-09-10:
Yeah I know about the flag, unfortunately it is not available.
[image]
Do you plan on building a recipe with P0829 ? Can I see somewhere the flags with which you compiled gcc, and maybe help on linking more of the stl?

ladislas, 2021-09-10:
We only provide support for avr-libc, wich is a subset of the C standard library.

https://www.nongnu.org/avr-libc/

C++ syntax does work but the STL is not available as is.

Alternatives do exist:

- https://community.platformio.org/t/c-stl-on-an-avr-platform/10351
- https://github.com/andysworkshop/avr-stl
- https://github.com/PhantomEmbedded/avr-stl
- https://www.etlcpp.com/ - my favorite

ladislas, 2021-09-10:
> Do you plan on building a recipe with P0829 freestanding proposal?

No, but I'm happy to look at a PR if needed.

> Can I see somewhere the flags with which you compiled gcc, and maybe help on linking more of the stl?

Here: https://github.com/osx-cross/homebrew-avr/blob/master/Formula/avr-gcc%4010.rb

ladislas, 2021-09-10:
And also here: https://www.arduino.cc/reference/en/libraries/arduinostl/

SpyrosMourelatos, 2021-09-13:
Thank for the feedback @ladislas , I am aware of the alternatives though nothing is good enough as my final goal is to use some libraries written for c++17 that use P0829 freestanding stdlibc++ from arm-none-eabi-g++. So it is not that I just want the atomic behaviour I need the exact same api. I was hoping to find something as good as arm-none-eabi-g++ on avr but now I know that there are many technical difficulties such as this one [de-depricating volatile](http://www.open-std.org/jtc1/sc22/wg21/docs/papers/2021/p2327r0.pdf) to make a good avr-g++12  .
I ended up copying this project [libstc++ P0829](https://gitlab.com/avr-libstdcxx) which has done some quite minimal patches(see the commits on the gcc repository of Paul Bixen, they are really simple) to the official g++ just for the P0829 , I also compiled it with the flags from the dockerimage on the same repo and it seems to work fine with my atmega328p :
`../configure --prefix=$PREFIX --target=avr --enable-languages=c,c++ --disable-nls --disable-libssp --with-dwarf2 --with-newlib --disable-__cxa_atexit --disable-threads --disable-shared --enable-static --disable-sjlj-exceptions --enable-libstdcxx --enable-lto --disable-hosted-libstdcxx `
Unfortunately I don't know if it could be a good idea to patch all this things to your recipe as it would not be the "one and true g++" anymore so I don't know how could I help. I would suggest closing this issue with some unhappiness :(

ladislas, 2021-09-13:
We could make this optional with a patch. 
Of even a stand-alone formulae if it gains traction.

ladislas, 2021-10-05:
@SpyrosMourelatos did you manage to find a solution for your issue?

I'll close for now, feel free to reopen in the future if needed.

closed 2021-10-05 by ladislas