osx-cross/homebrew-avr #262: Default version

bartekpacia, 2022-02-05:
`brew install avr-gcc` by default installs `avr-gcc@9` Formula, but `avr-gcc@11` is the newest version. Why is that and will it be changed?

ladislas, 2022-02-05:
Last time I checked, Arduino was using GCC 7.3.0.

the default here is already 2 versions newer than what people expect from Arduino. 

On such small and resources constrained systems, changing the compiler can have negative impact that would break otherwise working project. 

GCC 9 has been out long enough for people feeling comfortable using it in embedded, that's why it is the default.

I'm not in favor in changing the default right now. I'd prefer to wait for GCC 9 to stop being supported. I'm also happy to reconsider if a lot of people are in favor of that.

But then, why 11 and not 10? That would be the careful choice.

bartekpacia, 2022-02-05:
Sure, your answer makes sense. I just thought that I'd ask because I didn't find a similar issue :)

ladislas, 2022-02-15:
I'm closing for now, feel free to reopen in the future ;à

closed 2022-02-15 by ladislas