osx-cross/homebrew-avr #264: avr-gcc@9 Cannot flash on macOS Monterrey

vmnavarro94, 2022-03-01:
Im trying to flash a .hex onto my corne keyboard and got this error

`    Connecting to programmer: .
    Found programmer: Id = "CATERIN"; type = S
        Software Version = 1.0; No Hardware Version given.
    Programmer supports auto addr increment.
    Programmer supports buffered memory access with buffersize=128 bytes.
    
    Programmer supports the following devices:
        Device code: 0x44
    
    avrdude: AVR device initialized and ready to accept instructions
    
    Reading | ################################################## | 100% 0.00s
    
    avrdude: Device signature = 0x1e9587 (probably m32u4)
    avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
    avrdude: erasing chip
    avrdude: reading input file "/Users/vnavarro/qmk_firmware/.build/crkbd_rev1_vnav.hex"
    avrdude: can't open input file /Users/vnavarro/qmk_firmware/.build/crkbd_rev1_vnav.hex: No such file or directory
    avrdude: read from file '/Users/vnavarro/qmk_firmware/.build/crkbd_rev1_vnav.hex' failed
    
    avrdude done.  Thank you.`

I search over the net and some people suggest to downgrade to avr-gcc@7 but when trying to install I get this:

```
╰─ brew install avr-gcc@7                                                                                                                                                                                          ─╯
Warning: No available formula with the name "avr-gcc@7". Did you mean avr-gcc@9, avr-gcc@8, avr-gcc@5, avr-gcc@11 or avr-gcc@10?
==> Searching for similarly named formulae...
These similarly named formulae were found:
avr-gcc@9                                  avr-gcc@8                                  avr-gcc@5                                  avr-gcc@11                                 avr-gcc@10
To install one of them, run (for example):
  brew install avr-gcc@9
==> Searching for a previously deleted formula (in the last month)...
Error: No previously deleted formula found.
==> Searching taps on GitHub...
Error: No formulae found in taps.
```

ladislas, 2022-03-01:
@fauxpark any idea?

ladislas, 2022-03-01:
@vmnavarro94 regarding avr-gcc@7, it was removed in 7f3b60200a5981e6bd6f25aa950f064d419d7cdd

If you need it back, you can open a PR adding the formula back, see: https://github.com/osx-cross/homebrew-avr/blob/372486e66c955ec6ab029836cc792a8930dada64/Formula/avr-gcc%407.rb

fauxpark, 2022-03-01:
Nothing to do with avr-gcc or osx-cross. My guess would be something code-related.
Pleas file an issue at https://github.com/qmk/qmk_firmware.

closed 2022-05-06 by ladislas