osx-cross/homebrew-avr #265: Dutifully reporting error I have no understanding of

johnmee, 2022-05-05:
Just doing a routine `brew upgrade` and the fact that it suggests "if reporting this issue...":

```
==> Upgrading osx-cross/avr/avr-binutils
  2.36.1 -> 2.38 

==> Patching
==> Applying avr-size.patch
patching file binutils/size.c
Hunk #6 succeeded at 395 (offset -2 lines).
Hunk #7 succeeded at 414 (offset -2 lines).
Hunk #8 succeeded at 464 (offset -2 lines).
Hunk #9 succeeded at 909 (offset -2 lines).
Hunk #10 succeeded at 1008 (offset -2 lines).
==> Applying avr-binutils-elf-bfd-gdb-fix.patch
patching file bfd/elf-bfd.h
Hunk #1 succeeded at 21 with fuzz 2.
==> ../configure --prefix=/home/linuxbrew/.linuxbrew/Cellar/avr-binutils/2.38 --libdir=/home/linuxbrew/.linuxbrew/Cellar/avr-binutils/2.38/lib/avr --infodir=/hom
==> make
Last 15 lines from /home/john/.cache/Homebrew/Logs/avr-binutils/02.make:
         <http://www.gnu.org/software/make/>
configure: WARNING:
*** Makeinfo is too old. Info documentation will not be built.
checking for expect... no
checking for Tcl supporting try/catch... no
checking for linker versioning flags... -Wl,--version-script='../../libctf/libctf.ver'
configure: updating cache ./config.cache
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating config.h
config.status: executing depfiles commands
config.status: executing libtool commands
make[1]: Leaving directory '/tmp/avr-binutils-20220506-805661-1t4ucop/binutils-2.38/build'
make: *** [Makefile:999: all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
  https://github.com/osx-cross/homebrew-avr/issues
```

Ubuntu 21.10

ladislas, 2022-05-06:
We do not officially support Ubuntu. I don't have access to a machine right now so I can't really help.

johnmee, 2022-05-06:
Blame QMK...
https://docs.qmk.fm/#/cli?id=install-using-homebrew
Even if they do say "some linux"

Thx... loving your diligence.

closed 2022-05-06 by johnmee