osx-cross/homebrew-avr #266: Support for newer MCUs

lukeadair, 2022-05-21:
The version of avr-libc (2.0.0) that is currently used does not support newer MCUs like the attiny1616, but they recently released 2.1.0 that does support them.

Please consider updating this to use the latest version found here:
https://github.com/avrdudes/avr-libc/blob/main/NEWS

ladislas, 2022-05-21:
Ah! thanks for letting me know, let's see if it works :) https://github.com/osx-cross/homebrew-avr/pull/267

lukeadair, 2022-05-21:
Awesome! That was fast. Looks like it failed though.

ladislas, 2022-05-29:
@lukeadair I've tried with gcc@9 & gcc@10 with no success. I don't have time to investigate locally.

If you can give it a try and report any findings, it would be greatly appreciated.

ladislas, 2022-06-12:
This has been fixed thanks to @fauxpark 

see #267 & #274

closed 2022-06-12 by ladislas