osx-cross/homebrew-avr #270: Catalina bottles not being built

fauxpark, 2022-05-23:
`macos-latest` has been updated to macOS 11 (https://github.com/actions/virtual-environments/pull/4048). As a result, only the Big Sur bottle for avr-binutils 2.38 was built.

fauxpark, 2022-05-28:
Hi @ladislas, any update on this? Sorry to bother.

ladislas, 2022-05-28:
I'm afraid there's nothing I/we can do about that. 
Unless you have a machine still running Catalina that you want to setup for CI, Catalina users will need to build the packages from source.

That being said, Big Sur is compatible with "very old" computers (see https://support.apple.com/en-us/HT211238) and Catalina is not supported anymore even for security fixes. I would encourage users to upgrade.

ladislas, 2022-05-28:
Wait, it seems that Catalina is still available! (?)

https://github.com/actions/virtual-environments

fauxpark, 2022-05-28:
Yes, that's what I was about to post. They've only changed `macos-latest` to point to Big Sur; Catalina hasn't been dropped (yet).

ladislas, 2022-05-28:
fixed.

commit 9d8f3126 (2022-05-28): :green_heart: (actions): Rename macOS images

fix #270
PR #271 (2022-05-28): :green_heart: (actions): Rename macOS images
fix #270
closed 2022-05-28 by ladislas