osx-cross/homebrew-avr #277: macOS 13

laris, 2022-06-15:
`==> Installing avr-binutils from osx-cross/avr
==> Patching
==> Applying avr-size.patch
patching file 'binutils/size.c'
==> Applying avr-binutils-elf-bfd-gdb-fix.patch
patching file 'bfd/elf-bfd.h'
==> ../configure --prefix=/opt/homebrew/Cellar/avr-binutils/2.38_1 --libdir=/opt/homebrew/Cellar/avr-binutils/2.38_1/lib/avr --in
==> make
Last 15 lines from /Users/laris/Library/Logs/Homebrew/avr-binutils/02.make:
         want to install GNU make:
         <http://www.gnu.org/software/make/>
configure: WARNING:
*** Makeinfo is too old. Info documentation will not be built.
checking for expect... /usr/bin/expect
checking for Tcl supporting try/catch... no
checking for linker versioning flags... -export-symbols-regex ctf_.*
configure: updating cache ./config.cache
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating config.h
config.status: executing depfiles commands
config.status: executing libtool commands
make: *** [all] Error 2

Do not report this issue to Homebrew/brew or Homebrew/core!

Error: You are using macOS 13.
We do not provide support for this pre-release version.
You will encounter build failures with some formulae.
Please create pull requests instead of asking for help on Homebrew's GitHub,
Twitter or any other official channels. You are responsible for resolving
any issues you experience while you are running this
pre-release version.`

ladislas, 2022-06-15:
well the answer is written in your message:

> Do not report this issue to Homebrew/brew or Homebrew/core!
>
>
> Error: You are using macOS 13.
> We do not provide support for this pre-release version.
> You will encounter build failures with some [...]

closed 2022-06-15 by ladislas

futuun, 2022-07-09:
I just got this (macos 13, xcode 14 with CLI tools installed), and for me `makeinfo` was missing (you can get more details from running brew commands with `-vd`).
Installing `texinfo` and adding `depends_on "texinfo" => :build` to formula (can be done with `brew edit avr-binutils`) makes it work.

ladislas, 2022-07-09:
As said before, and as main homebrew does, we do not support pre-release versions.

I'm open to a PR to look at what CI says but we will not merge until Ventura is officially released.

commit 0f0a6cd3 (2022-10-25): avr-binutils: rebuild

Fix Ventura build by requiring texinfo

Per #277, building fails on macOS Ventura with a missing `makeinfo`
error. This commit adds `texinfo` as a build requirement.

Closes #277

Closes #283.