osx-cross/homebrew-avr #280: avr-gcc@11 fails to build

alexisvl, 2022-07-22:
avr-gcc@11 fails to build:

```
==> make BOOT_LDFLAGS=-Wl,-headerpad_max_install_names
Last 15 lines from /Users/alex/Library/Logs/Homebrew/avr-gcc@11/02.make:
../../gcc/config/avr/avr.c:10243:36: error: assigning to 'unnamed_section_callback' (aka 'void (*)(const char *)') from incompatible type 'void (const void *)': type mismatch at 1st parameter ('const char *' vs 'const void *')
  data_section->unnamed.callback = avr_output_data_section_asm_op;
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../gcc/config/avr/avr.c:10244:35: error: assigning to 'unnamed_section_callback' (aka 'void (*)(const char *)') from incompatible type 'void (const void *)': type mismatch at 1st parameter ('const char *' vs 'const void *')
  bss_section->unnamed.callback = avr_output_bss_section_asm_op;
                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
clang++ -std=gnu++11  -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-strict-aliasing -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wno-error=format-diag    -DHAVE_CONFIG_H -I. -Ianalyzer -I../../gcc -I../../gcc/analyzer -I../../gcc/../include -I../../gcc/../libcpp/include -I../../gcc/../libcody  -I../../gcc/../libdecnumber -I../../gcc/../libdecnumber/dpd -I../libdecnumber -I../../gcc/../libbacktrace   -o analyzer/analyzer-selftests.o -MT analyzer/analyzer-selftests.o -MMD -MP -MF analyzer/.deps/analyzer-selftests.TPo ../../gcc/analyzer/analyzer-selftests.cc
clang++ -std=gnu++11  -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-strict-aliasing -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wno-error=format-diag    -DHAVE_CONFIG_H -I. -Ianalyzer -I../../gcc -I../../gcc/analyzer -I../../gcc/../include -I../../gcc/../libcpp/include -I../../gcc/../libcody  -I../../gcc/../libdecnumber -I../../gcc/../libdecnumber/dpd -I../libdecnumber -I../../gcc/../libbacktrace   -o analyzer/bar-chart.o -MT analyzer/bar-chart.o -MMD -MP -MF analyzer/.deps/bar-chart.TPo ../../gcc/analyzer/bar-chart.cc
clang++ -std=gnu++11  -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-strict-aliasing -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wno-error=format-diag    -DHAVE_CONFIG_H -I. -Ianalyzer -I../../gcc -I../../gcc/analyzer -I../../gcc/../include -I../../gcc/../libcpp/include -I../../gcc/../libcody  -I../../gcc/../libdecnumber -I../../gcc/../libdecnumber/dpd -I../libdecnumber -I../../gcc/../libbacktrace   -o analyzer/call-string.o -MT analyzer/call-string.o -MMD -MP -MF analyzer/.deps/call-string.TPo ../../gcc/analyzer/call-string.cc
5 errors generated.
make[2]: *** [avr.o] Error 1
```

[02.make](https://github.com/osx-cross/homebrew-avr/files/9164102/02.make.txt)

Full command output:

[error.txt](https://github.com/osx-cross/homebrew-avr/files/9164098/error.txt)

macOS 12.5

clang 13.1.6

Let me know what else I can provide here - I'm pretty new to homebrew

45gfg9, 2022-08-18:
Same problem on a M1 Max MacBook.
```
==> Installing osx-cross/avr/avr-gcc@11
==> Patching
==> Applying gcc-11.3.0-arm.diff
patching file Makefile.def
patching file Makefile.in
patching file Makefile.tpl
patching file c++tools/config.h.in
patching file c++tools/configure
patching file c++tools/configure.ac
patching file c++tools/resolver.cc
patching file c++tools/resolver.h
patching file c++tools/server.cc
patching file config/gcc-plugin.m4
patching file config/mh-darwin
patching file configure
patching file configure.ac
patching file fixincludes/configure
patching file gcc/Makefile.in
patching file gcc/aclocal.m4
patching file gcc/ada/gcc-interface/Make-lang.in
patching file gcc/ada/gcc-interface/Makefile.in
patching file gcc/ada/gcc-interface/decl.c
patching file gcc/ada/sigtramp-ios.c
patching file gcc/builtins.def
patching file gcc/c-family/c-attribs.c
patching file gcc/c-family/c-opts.c
patching file gcc/c-family/c-pch.c
patching file gcc/c/c-decl.c
patching file gcc/c/c-tree.h
patching file gcc/c/c-typeck.c
patching file gcc/calls.c
patching file gcc/calls.h
patching file gcc/common.opt
patching file gcc/config.gcc
patching file gcc/config.host
patching file gcc/config.in
patching file gcc/config/aarch64/aarch64-builtins.c
patching file gcc/config/aarch64/aarch64-c.c
patching file gcc/config/aarch64/aarch64-protos.h
patching file gcc/config/aarch64/aarch64-sve-builtins.cc
patching file gcc/config/aarch64/aarch64.c
patching file gcc/config/aarch64/aarch64.h
patching file gcc/config/aarch64/aarch64.md
patching file gcc/config/aarch64/aarch64.opt
patching file gcc/config/aarch64/constraints.md
patching file gcc/config/aarch64/darwin.h
patching file gcc/config/aarch64/darwinpcs.md
patching file gcc/config/aarch64/falkor-tag-collision-avoidance.c
patching file gcc/config/aarch64/host-aarch64-darwin.c
patching file gcc/config/aarch64/predicates.md
patching file gcc/config/aarch64/t-aarch64-darwin
patching file gcc/config/aarch64/x-darwin
patching file gcc/config/avr/avr.c
patching file gcc/config/darwin-driver.c
patching file gcc/config/darwin.c
patching file gcc/config/darwin.h
patching file gcc/config/darwin.opt
patching file gcc/config/host-darwin.c
patching file gcc/config/host-darwin.h
patching file gcc/config/host-hpux.c
patching file gcc/config/host-linux.c
patching file gcc/config/host-netbsd.c
patching file gcc/config/host-openbsd.c
patching file gcc/config/host-solaris.c
patching file gcc/config/i386/darwin.h
patching file gcc/config/i386/host-mingw32.c
patching file gcc/config/i386/i386.c
patching file gcc/config/i386/i386.h
[... 336 lines cut ...]
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../gcc/config/avr/avr.c:10244:35: error: assigning to 'unnamed_section_callback' (aka 'void (*)(const char *)') from incompatible type 'void (const void *)': type mismatch at 1st parameter ('const char *' vs 'const void *')
  bss_section->unnamed.callback = avr_output_bss_section_asm_op;
                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
clang++ -std=gnu++11  -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-strict-aliasing -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wno-error=format-diag    -DHAVE_CONFIG_H -I. -Ianalyzer -I../../gcc -I../../gcc/analyzer -I../../gcc/../include -I../../gcc/../libcpp/include -I../../gcc/../libcody  -I../../gcc/../libdecnumber -I../../gcc/../libdecnumber/dpd -I../libdecnumber -I../../gcc/../libbacktrace   -o analyzer/complexity.o -MT analyzer/complexity.o -MMD -MP -MF analyzer/.deps/complexity.TPo ../../gcc/analyzer/complexity.cc
5 errors generated.
make[2]: *** [avr.o] Error 1
make[2]: *** Waiting for unfinished jobs....
rm gcc.pod
make[1]: *** [all-gcc] Error 2
make: *** [all] Error 2

Do not report this issue to Homebrew/brew or Homebrew/core!

These open issues may also help:
avr-gcc@11 fails to build https://github.com/osx-cross/homebrew-avr/issues/280
```
Same version macOS and clang.
```shell
$ clang -v
Apple clang version 13.1.6 (clang-1316.0.21.2.5)
Target: arm64-apple-darwin21.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

jmechnich, 2022-10-08:
I was able to fix it for me by applying this patch: https://gist.github.com/jmechnich/5a31cf9c2ae8336109f48abd3e5488a0

You can `brew edit avr-gcc@11` and add the following to the formula locally:
```diff
diff --git a/Formula/avr-gcc@11.rb b/Formula/avr-gcc@11.rb
index e8a0160..9e06fe9 100644
--- a/Formula/avr-gcc@11.rb
+++ b/Formula/avr-gcc@11.rb
@@ -65,6 +65,10 @@ class AvrGccAT11 < Formula
       url "https://raw.githubusercontent.com/Homebrew/formula-patches/22dec3fc/gcc/gcc-11.3.0-arm.diff"
       sha256 "e02006b7ec917cc1390645d95735a6a866caed0dfe506d5bef742f7862cab218"
     end
+    patch do
+      url "https://gist.githubusercontent.com/jmechnich/5a31cf9c2ae8336109f48abd3e5488a0/raw/e42700b3519a31a06dbdc6095a11e6d101fb75f3/avr-gcc@11.arm.patch"
+      sha256 "79232a7dcbe71bcf4a0cc52e84cd509553d7b40b887771eec06ac340f5f502f6"
+    end
   end
 
   def version_suffix
```

Not really sure how / where I could submit this...

Patryk27, 2023-02-17:
fwiw, this has been already fixed on GCC's side (https://github.com/gcc-mirror/gcc/commit/4dc6d19222581c77a174d44d97507d234fb7e39b) 🙂

ladislas, 2023-02-17:
@Patryk27 could you make a PR? :)

jmechnich, 2023-02-18:
Judging from the commit message in the link above, It looks like this has only been fixed in gcc 12.1 and up though.

This still has to be backported to 11.x for GCC (and there does not seem to be an easy way to submit a PR to them).

Patryk27, 2023-02-18:
Ah, right, yes - it's GCC 12.

jmechnich, 2023-02-19:
As nothing happened upstream since my [post](https://github.com/osx-cross/homebrew-avr/issues/280#issuecomment-1272381808) from Oct 2022, I have now submitted [PR](https://github.com/osx-cross/homebrew-avr/pull/287) to this repo.

I have no idea if it would have been more appropriate to put the patch into [Homebrew/formula-patches](https://github.com/Homebrew/formula-patches) but there you go...

jmechnich, 2023-02-27:
The most recent commits have fixed this issue for me (on macOS 13 arm64). Thanks @ladislas !

ladislas, 2023-02-27:
glad to hear! I'll close the issue for now, feel free to open a new one if needed.

closed 2023-02-27 by ladislas