osx-cross/homebrew-avr #319: simavr1.6 Makefile bug checking for avr-gcc installation

Jim-Carroll, 2023-11-11:
The current version of the simavr1.6 Makefile.common checks if avr-gcc is installed by looking for a folder called `avr-gcc` in the homebrew Cellar. This did not work on my system (macOS Ventura 13.5). Instead, I have two folders in the Cellar called `avr-gcc@9` and `avr-gcc@12`.

To workaround this issue, I manually created a folder called `avr-gcc` in the Cellar. This allowed the installation to proceed and complete successfully.

```
# workaround
mkdir $(brew --prefix)/Cellar/avr-gcc
# now this works
brew install simavr
```

Looks like the new version of the Makefile fixes this by using a glob pattern, so I guess this cask is just outdated. However, the recommendation to `brew install avr-libc` is incorrect. No such package exists. I believe the error message should say `brew install avr-gcc` instead.

https://github.com/buserror/simavr/blob/master/Makefile.common#L69C4-L69C4

gatk555, 2023-12-28:
Should this not be opened at buserror/simavr?

Jim-Carroll, 2023-12-28:
No. It has already been fixed upstream. This repo is just using an older version.

https://github.com/buserror/simavr/commit/9f1014a7ae80a0338d10a6152f2fee0e5a6e410d

gatk555, 2023-12-29:
My mistake! But there is nothing on the surface to indicate that a simavr build is here.

Thanks, G.

Jim-Carroll, 2023-12-29:
This formula uses simavr v1.6: https://github.com/osx-cross/homebrew-avr/blob/main/Formula/simavr.rb

That version of simavr has the aforementioned bug.

kaidokert, 2025-07-11:
Should be fixed by #359

closed 2025-11-14 by ladislas