osx-cross/homebrew-avr #320: Abort trap: 6 signal terminated program cc1

rwijtvliet, 2023-11-15:
Error when compiling `qmk` (`qmk compile ..`):

```bash
(...)

Compiling: .build/obj_keyboardio_atreus_rwijtvliet/src/default_keyboard.c                          dyld[50335]: Library not loaded: /opt/homebrew/opt/mpfr/lib/libmpfr.6.dylib
  Referenced from: <B938679C-CFBC-3146-9708-0FD82ABDEBD4> /opt/homebrew/Cellar/avr-gcc@8/8.5.0_1/libexec/gcc/avr/8.5.0/cc1
  Reason: tried: '/opt/homebrew/opt/mpfr/lib/libmpfr.6.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/opt/mpfr/lib/libmpfr.6.dylib' (no such file), '/opt/homebrew/opt/mpfr/lib/libmpfr.6.dylib' (no such file)
avr-gcc: internal compiler error: Abort trap: 6 signal terminated program cc1
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://github.com/osx-cross/homebrew-avr/issues> for instructions.
 [ERRORS]
 |
 |
 |
gmake[1]: *** [builddefs/common_rules.mk:361: .build/obj_keyboardio_atreus_rwijtvliet/.build/obj_keyboardio_atreus_rwijtvliet/src/default_keyboard.o] Error 1
if [ -f .build/error_occurred ]; then printf "\033[31;01mMake finished with errors\n\033[0m" & exit 1; fi;
gmake: *** [Makefile:392: keyboardio/atreus:rwijtvliet] Error 1
Make finished with errors
```

My machine: macos m2 sonoma 14.1

Output of `brew list`:

```bash
❯ brew list
==> Formulae
aom			gnu-indent		libtiff			pcre2
arm-none-eabi-binutils	gnu-sed			libtool			pillow
arm-none-eabi-gcc@8	gnu-tar			libunistring		pixman
autoconf		gnutls			libusb			pkg-config
automake		graphite2		libusb-compat		python@3.11
avr-binutils		grep			libuv			qmk
avr-gcc@8		harfbuzz		libvmaf			readline
avrdude			hid_bootloader_cli	libvterm		ripgrep
bash			hidapi			libx11			screenresolution
boost			highway			libxau			shared-mime-info
bootloadhid		icu4c			libxcb			sketchybar
brotli			imagemagick		libxdmcp		skhd
c-ares			imath			libxext			sqlite
ca-certificates		isl			libxrender		starship
cairo			jasper			little-cms2		switchaudio-osx
clang-format		jbig2dec		luajit			syncthing
cliclick		jpeg-turbo		luv			tcl-tk
confuse			jpeg-xl			lz4			teensy_loader_cli
coreutils		jq			lzo			texinfo
dfu-programmer		libde265		m4			tree
dfu-util		libevent		make			tree-sitter
dockutil		libftdi			mdloader		unbound
fd			libheif			mpdecimal		unibilium
findutils		libidn			mpfr			wallpaper
fontconfig		libidn2			msgpack			webp
freetype		libimagequant		neofetch		wget
fribidi			liblqr			neovim			wireguard-go
gawk			libmpc			nettle			wireguard-tools
gettext			libnghttp2		node			x265
gh			libomp			oniguruma		xorgproto
ghostscript		libpng			openexr			xz
giflib			libraqm			openjdk			yabai
glib			libraw			openjpeg		zstd
gmp			libtasn1		openssl@3
gnu-getopt		libtermkey		p11-kit

==> Casks
alacritty		inkscape		sf-symbols		visual-studio-code
commander-one		kawa			spotify			vscodium
espanso			miniconda		topnotch
freecad			raycast			unnaturalscrollwheels
```

Not sure what else I can tell you. If you want me to check something, let me know!

ladislas, 2023-11-15:
I'll ping @fauxpark here :)

fauxpark, 2023-11-15:
Run `brew doctor`, if all is well run `brew reinstall mpfr`. Seems like you're missing the dylib from that package; no idea how that could have happened.

rwijtvliet, 2023-11-15:
That fixed it! Many thanks!

closed 2023-11-15 by rwijtvliet