osx-cross/homebrew-avr #326: Installation fails due to C++ error (?)

jose-elias-alvarez, 2024-03-08:
I'm attempting to install `avr-gcc` on an M1 MacBook Pro running macOS Sonoma 14.4 and am getting the following output, after which the command errors out:

```
==> Fetching osx-cross/avr/avr-gcc@9
==> Downloading https://gist.githubusercontent.com/DavidEGrayson/88bceb3f4e62f45725ecbb9248366300/raw/c1f515
Already downloaded: /Users/jose/Library/Caches/Homebrew/downloads/def654fc009e1286443dbc4e1baa98eeeaee383d419d8a1d15200ab895520379--gcc-11-arm-darwin.patch
==> Downloading https://download.savannah.gnu.org/releases/avr-libc/avr-libc-2.1.0.tar.bz2
Already downloaded: /Users/jose/Library/Caches/Homebrew/downloads/82989e2eb60e70483e667482e25d205685a09019a1c993b99a467ad14d3bb2d0--avr-libc-2.1.0.tar.bz2
==> Downloading https://ftp.gnu.org/gnu/gcc/gcc-9.4.0/gcc-9.4.0.tar.xz
Already downloaded: /Users/jose/Library/Caches/Homebrew/downloads/573d147d9b95f1a25da4ceee90b6232279397864eb5fa0b2802ff97ef8a5d8ed--gcc-9.4.0.tar.xz
==> Installing avr-gcc@9 from osx-cross/avr

==> Patching
==> Applying gcc-11-arm-darwin.patch
patching file 'gcc/config/host-darwin.c'
==> ../configure --target=avr --prefix=/opt/homebrew/Cellar/avr-gcc@9/9.4.0 --libdir=/opt/homebrew/Cellar/av
==> make BOOT_LDFLAGS=-Wl,-headerpad_max_install_names
Last 15 lines from /Users/jose/Library/Logs/Homebrew/avr-gcc@9/02.make:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk/usr/include/c++/v1/__functional/boyer_moore_searcher.h:27:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk/usr/include/c++/v1/vector:321:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk/usr/include/c++/v1/__format/formatter_bool.h:20:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk/usr/include/c++/v1/__format/formatter_integral.h:32:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk/usr/include/c++/v1/locale:202:
/Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk/usr/include/c++/v1/__locale:811:8: error: expected ';' at end of declaration
iscntrl(_CharT __c, const locale& __loc)
       ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [gcov.o] Error 1
make[2]: *** Waiting for unfinished jobs....
rm gcc.pod
make[1]: *** [all-gcc] Error 2
make: *** [all] Error 2
```

I can only speculate as to the cause, as I'm not familiar with this library and am just attempting to install QMK, but let me know if I can provide any more info.

Jmarinis, 2024-03-09:
Reproduced the exact same error on Sonoma 14.4 on M1 Mac Mini.  Latest version of Xcode.

AndrewKovalenko, 2024-03-10:
Consistently reproducible issue on m2 Sonoma 14.4

Nibir, 2024-03-12:
reproducible issue on m1 Sonoma 14.4

Jmarinis, 2024-03-15:
Appears to be fixed in avr-gcc@9/9.4.0_1

jose-elias-alvarez, 2024-03-16:
Also working on my end, thanks!

closed 2024-03-16 by jose-elias-alvarez

fauxpark, 2024-03-17:
Technically this is not "fixed". The reason it's working is because there are now Apple Silicon bottles available for Sonoma (#324), avoiding the need to compile these formulae on your local machine. However the macOS SDK this new Sonoma runner uses is still at 14.2, not 14.4, which appears to be the one causing this issue.

When this inevitably gets updated future builds may fail, but at least the bottles should still be available.