osx-cross/homebrew-avr #339: Move to latest avr-libc (and new repo)

jeremyherbert, 2024-08-15:
Hi,

avr-libc has a new repository here: https://github.com/avrdudes/avr-libc

They also have a new release of avr-libc which supports a lot of the newer AVR parts (AVR128D, etc): https://github.com/avrdudes/avr-libc/releases/tag/avr-libc-2_2_1-release

I tested this new release by editing the formula file for `avr-gcc@14.rb` and it worked fine (I built for the new AVR128DB48 which is only supported in avr-libc 2.2+)

Could we please update this repo to include the latest avr-libc? I am happy to create a PR with the updated URLs/hashes, please let me know.

ladislas, 2024-08-15:
Yes please do :)

ladislas, 2024-11-25:
this has been mostly implemented. I'll close the issue for now.

closed 2024-11-25 by ladislas