osx-cross/homebrew-avr #342: avr-gcc: error: unrecognized command-line option '-arch'

theoparis, 2024-09-01:
I am trying to use cmake to compile with a toolchain file that sets the C compiler to avr-gcc, but something seems to be appending `-arch arm64` (maybe this is a cmake bug)... Any ideas?

Command:
```
cmake -B build -G Ninja -DCMAKE_TOOLCHAIN_FILE=toolchain.cmake
```

Build log:
```
    /opt/homebrew/opt/avr-gcc@14/bin/avr-gcc   -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.5.sdk -mmcu=atmega328p -Os -o CMakeFiles/cmTC_ada1d.dir/testCCompiler.c.o -c /Users/theo/Documents/src/bruh/avr-cmake/build/CMakeFiles/CMakeScratch/TryCompile-ttvhZ9/testCCompiler.c
    avr-gcc: error: unrecognized command-line option '-arch'
    ninja: build stopped: subcommand failed.
```

theoparis, 2024-09-01:
Update: I added `set(CMAKE_SYSTEM_PROCESSOR avr)` to my toolchain.cmake file and I got this
```
    /opt/homebrew/opt/avr-binutils/bin/avr-ld: Error: unable to disambiguate: -search_paths_first (did you mean --search_paths_first ?)
```

Building cmake from source with the following patch seems to fix it - maybe I should report this upstream for a better fix.
```diff
diff --git a/Modules/Platform/Darwin.cmake b/Modules/Platform/Darwin.cmake
index f334c5f5cd..43816a0060 100644
--- a/Modules/Platform/Darwin.cmake
+++ b/Modules/Platform/Darwin.cmake
@@ -56,7 +56,7 @@ foreach(lang C CXX OBJC OBJCXX)
   set(CMAKE_${lang}_OSX_CURRENT_VERSION_FLAG "-current_version ")
   set(CMAKE_${lang}_LINK_FLAGS "-Wl,-headerpad_max_install_names")
 
-  if(HAVE_FLAG_SEARCH_PATHS_FIRST)
+  if(HAVE_FLAG_SEARCH_PATHS_FIRST AND NOT CMAKE_C_COMPILER_TARGET)
     set(CMAKE_${lang}_LINK_FLAGS "-Wl,-search_paths_first ${CMAKE_${lang}_LINK_FLAGS}")
   endif()
 

```

ladislas, 2024-09-01:
this repo is not related to cmake. you can use cmake or make or bazel or whatever build system you want to use avr-gcc.

you should not build cmake from source, there is absolutely no reason to do it, just install it with homebrew like you did for avr-gcc with `brew upgrade && brew install cmake`

you should be familiar with cmake before starting to use it. I recommend:

- https://cliutils.gitlab.io/modern-cmake/README.html
- https://gist.github.com/mbinna/c61dbb39bca0e4fb7d1f73b0d66a4fd1

Then don't write your CMakeLists.txt yourself, start by using one provided by people with more knowledge and experience and that are known to work for your target architecture.

Just google gives me a few that I have tried but should work. From the quicklook, they are intended for expert users.

- https://github.com/mkleemann/cmake-avr
- https://os.everydaydev.io/docs/avr-cmake-template

Not sure why you need CMake, but if you're okay with Make, I would recommend https://github.com/sudar/Arduino-Makefile

I also have a template using it which works great https://github.com/ladislas/Bare-Arduino-Project

I'll close the issue. If you have trouble with this formula, feel free to open new issues.

closed 2024-09-01 by ladislas