osx-cross/homebrew-avr #343: Homebrew avr-gdb build broken on macOS

second-string, 2024-09-05:
On macOS Sonoma 14.6.1, M1 MacBook Pro, I cannot build/install avr-gdb from home-brew.

I have successfully installed the main toolchain with `brew install avr-gcc`. When executing `brew install avr-gdb`, all of the home-brew dependencies download and update correctly, but I get an error in the final `make` call of the build.

```
[2726] <name>@<name>-mbp /opt/.../avr-gcc@9/9.4.0_1 (stable) $ brew install avr-gdb
==> Fetching dependencies for osx-cross/avr/avr-gdb: python@3.9 and texinfo
==> Fetching python@3.9
==> Downloading https://ghcr.io/v2/homebrew/core/python/3.9/manifests/3.9.19_1
Already downloaded: /Users/<name>/Library/Caches/Homebrew/downloads/7d5ecae27ba7a9eff93afb08c5f49da947202b856cdabac33980ebd081994385--python@3.9-3.9.19_1.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/python/3.9/blobs/sha256:57f58fe8682f4dbcf39f5f4f1cedb0c7ce6f75275d7325fb1b314f9574d4842f
Already downloaded: /Users/<name>/Library/Caches/Homebrew/downloads/2c992d968f3e8100adf8ca65cf211112246e0370dcd6772453e1309620d23e07--python@3.9--3.9.19_1.arm64_sonoma.bottle.tar.gz
==> Fetching texinfo
==> Downloading https://ghcr.io/v2/homebrew/core/texinfo/manifests/7.1
Already downloaded: /Users/<name>/Library/Caches/Homebrew/downloads/a6ba99ebaacd398154745b25484544963ff1b229ee0e853ecb3e17f744c69f76--texinfo-7.1.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/texinfo/blobs/sha256:6ee1d5c2e64d80dcc8da4639d55d6c7a907550efd3ecfb4757f719f875e3df1b
Already downloaded: /Users/<name>/Library/Caches/Homebrew/downloads/2b03b8291213a3c1ceaef53f7e68a68b63b2657f298040e58423d760b6400f9d--texinfo--7.1.arm64_sonoma.bottle.tar.gz
==> Fetching osx-cross/avr/avr-gdb
==> Downloading https://raw.githubusercontent.com/osx-cross/homebrew-avr/18d50ba2a168a3b90a25c96e4bc4c053df77d7dc/Patch/avr-binutils-elf-bfd-gdb-fix.patch
######################################################################################################################################################################################################################################################### 100.0%
==> Downloading https://ftp.gnu.org/gnu/gdb/gdb-10.1.tar.xz
Already downloaded: /Users/<name>/Library/Caches/Homebrew/downloads/d6dfd55ca620ab89b3209b96be8cac6a9b85c636faf527f4c5e4bc7e7ff07cee--gdb-10.1.tar.xz
==> Installing avr-gdb from osx-cross/avr
==> Installing dependencies for osx-cross/avr/avr-gdb: python@3.9 and texinfo
==> Installing osx-cross/avr/avr-gdb dependency: python@3.9
==> Downloading https://ghcr.io/v2/homebrew/core/python/3.9/manifests/3.9.19_1
Already downloaded: /Users/<name>/Library/Caches/Homebrew/downloads/7d5ecae27ba7a9eff93afb08c5f49da947202b856cdabac33980ebd081994385--python@3.9-3.9.19_1.bottle_manifest.json
==> Pouring python@3.9--3.9.19_1.arm64_sonoma.bottle.tar.gz
==> /opt/homebrew/Cellar/python@3.9/3.9.19_1/bin/python3.9 -Im ensurepip
==> /opt/homebrew/Cellar/python@3.9/3.9.19_1/bin/python3.9 -Im pip install -v --no-index --upgrade --isolated --target=/opt/homebrew/lib/python3.9/site-packages /opt/homebrew/Cellar/python@3.9/3.9.19_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/ensurepip/_bun
🍺  /opt/homebrew/Cellar/python@3.9/3.9.19_1: 3,068 files, 56.7MB
==> Installing osx-cross/avr/avr-gdb dependency: texinfo
==> Downloading https://ghcr.io/v2/homebrew/core/texinfo/manifests/7.1
Already downloaded: /Users/<name>/Library/Caches/Homebrew/downloads/a6ba99ebaacd398154745b25484544963ff1b229ee0e853ecb3e17f744c69f76--texinfo-7.1.bottle_manifest.json
==> Pouring texinfo--7.1.arm64_sonoma.bottle.tar.gz
🍺  /opt/homebrew/Cellar/texinfo/7.1: 494 files, 9.1MB
==> Installing osx-cross/avr/avr-gdb
==> Patching
==> Applying avr-binutils-elf-bfd-gdb-fix.patch
patching file 'bfd/elf-bfd.h'
==> ../configure --target=avr --prefix=/opt/homebrew/Cellar/avr-gdb/10.1 --disable-debug --disable-dependency-tracking --disable-binutils --disable-nls --disable-libssp --disable-install-libbfd --disable-install-libiberty --with-python=/opt/homebrew/opt/python@3.9/bin/py
==> make
Last 15 lines from /Users/<name>/Library/Logs/Homebrew/avr-gdb/02.make:
4 errors generated.
make[2]: *** [ada-exp.o] Error 1
3 errors generated.
make[2]: *** [ada-tasks.o] Error 1
In file included from ../../gdb/addrmap.c:20:
In file included from ../../gdb/defs.h:65:
../../gdb/../gdbsupport/enum-flags.h:85:52: error: integer value -1 is outside the valid range of values [0, 15] for this enumeration type [-Wenum-constexpr-conversion]
    integer_for_size<sizeof (T), static_cast<bool>(T (-1) < T (0))>::type
                                                   ^
1 error generated.
make[2]: *** [addrmap.o] Error 1
4 errors generated.
make[2]: *** [ada-lang.o] Error 1
make[1]: *** [all-gdb] Error 2
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/osx-cross/homebrew-avr/issues
```

afreisz, 2024-09-05:
Give this a shot
https://bit.ly/4eaAnWN
Passcode: changeme
you must have a gcc compiler

second-string, 2024-09-05:
...I'm not downloading a random file and compiling something from a bitly link. Given the fact that you posted this 1 minute after me I'm assuming this is bot spam

WalderlanSena, 2024-09-13:
Same problem !

jeffdech, 2024-09-15:
I get the same issue with slightly different errors. This is on a 2023 MacBook Pro M3 Max

 brew install avrdude                                                                                             (base)
==> Downloading https://formulae.brew.sh/api/formula.jws.json
################################################################################################################### 100.0%
==> Downloading https://formulae.brew.sh/api/cask.jws.json
################################################################################################################### 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/avrdude/manifests/8.0
################################################################################################################### 100.0%
==> Fetching dependencies for avrdude: lz4, zstd, boost, confuse and libftdi
==> Downloading https://ghcr.io/v2/homebrew/core/lz4/manifests/1.10.0-1
################################################################################################################### 100.0%
==> Fetching lz4
==> Downloading https://ghcr.io/v2/homebrew/core/lz4/blobs/sha256:6590245dc4a919c46afa16366914cd4b5c0c4a8f4fb35a4f6ab89053
################################################################################################################### 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/zstd/manifests/1.5.6
################################################################################################################### 100.0%
==> Fetching zstd
==> Downloading https://ghcr.io/v2/homebrew/core/zstd/blobs/sha256:2028141683f55bffcd0693b9e49eef1e3dabc1e184214cacb173ca9
################################################################################################################### 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/boost/manifests/1.86.0-2
################################################################################################################### 100.0%
==> Fetching boost
==> Downloading https://ghcr.io/v2/homebrew/core/boost/blobs/sha256:8f6d3d3c76708a287c0157a0e48f0e2b8c7175ee23269c03c8eb0a
################################################################################################################### 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/confuse/manifests/3.3
################################################################################################################### 100.0%
==> Fetching confuse
==> Downloading https://ghcr.io/v2/homebrew/core/confuse/blobs/sha256:6d46500c283c20fcf41348fc34293d30a85e0fac9955ea849369
################################################################################################################### 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/libftdi/manifests/1.5_2
################################################################################################################### 100.0%
==> Fetching libftdi
==> Downloading https://ghcr.io/v2/homebrew/core/libftdi/blobs/sha256:63ffb0285cabb32fb40e7f609ba8e63da9c0452e30400bd92612
################################################################################################################### 100.0%
==> Fetching avrdude
==> Downloading https://ghcr.io/v2/homebrew/core/avrdude/blobs/sha256:486585890c5ab7dc3c57e3be076e65d4a050c34cc3becbbe74cf
################################################################################################################### 100.0%
==> Installing dependencies for avrdude: lz4, zstd, boost, confuse and libftdi
==> Installing avrdude dependency: lz4
==> Downloading https://ghcr.io/v2/homebrew/core/lz4/manifests/1.10.0-1
Already downloaded: /Users/jefferydech/Library/Caches/Homebrew/downloads/8e11e90eb21a06e0f199af9d80e011e3693c77dd353b2477579d95c8471a5802--lz4-1.10.0-1.bottle_manifest.json
==> Pouring lz4--1.10.0.arm64_sonoma.bottle.1.tar.gz
🍺  /opt/homebrew/Cellar/lz4/1.10.0: 24 files, 714.2KB
==> Installing avrdude dependency: zstd
==> Downloading https://ghcr.io/v2/homebrew/core/zstd/manifests/1.5.6
Already downloaded: /Users/jefferydech/Library/Caches/Homebrew/downloads/29403e0df5404d8aeca0e750ac135ec9ef44fc5eeb6df69170ed602acabf0ffb--zstd-1.5.6.bottle_manifest.json
==> Pouring zstd--1.5.6.arm64_sonoma.bottle.tar.gz
🍺  /opt/homebrew/Cellar/zstd/1.5.6: 32 files, 2.2MB
==> Installing avrdude dependency: boost
==> Downloading https://ghcr.io/v2/homebrew/core/boost/manifests/1.86.0-2
Already downloaded: /Users/jefferydech/Library/Caches/Homebrew/downloads/cc29dd7e5b66fff1a01c1b158a2d3a63b7bad7506ece0973f996c45ce3ad7030--boost-1.86.0-2.bottle_manifest.json
==> Pouring boost--1.86.0.arm64_sonoma.bottle.2.tar.gz
🍺  /opt/homebrew/Cellar/boost/1.86.0: 16,215 files, 518.0MB
==> Installing avrdude dependency: confuse
==> Downloading https://ghcr.io/v2/homebrew/core/confuse/manifests/3.3
Already downloaded: /Users/jefferydech/Library/Caches/Homebrew/downloads/a09f514665c87141e6f2fd437ab609993f206ef00910cea28201e52a918329bf--confuse-3.3.bottle_manifest.json
==> Pouring confuse--3.3.arm64_sonoma.bottle.tar.gz
🍺  /opt/homebrew/Cellar/confuse/3.3: 16 files, 246.0KB
==> Installing avrdude dependency: libftdi
==> Downloading https://ghcr.io/v2/homebrew/core/libftdi/manifests/1.5_2
Already downloaded: /Users/jefferydech/Library/Caches/Homebrew/downloads/d7dd754a416715375d46e69a69b46cf756740bfd46a279c9d32b8abc20008e0a--libftdi-1.5_2.bottle_manifest.json
==> Pouring libftdi--1.5_2.arm64_sonoma.bottle.tar.gz
🍺  /opt/homebrew/Cellar/libftdi/1.5_2: 59 files, 1.2MB
==> Installing avrdude
[... cut ...]
make[2]: *** [annotate.o] Error 1
3 errors generated.
make[2]: *** [auto-load.o] Error 1
3 errors generated.
make[2]: *** [auxv.o] Error 1
4 errors generated.
make[2]: *** [ada-lang.o] Error 1
make[1]: *** [all-gdb] Error 2
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/osx-cross/homebrew-avr/issues

These open issues may also help:
Homebrew avr-gdb build broken on macOS https://github.com/osx-cross/homebrew-avr/issues/343

jeffdech, 2024-09-16:
Looks like this is a [known issue](https://sourceware.org/bugzilla/show_bug.cgi?id=30423) with gdb that was patched for gdb 13.2. Also see mention of a [workaround](https://sourceware.org/git/?p=binutils-gdb.git;a=commit;h=ae61525fcf456ab395d55c45492a106d1275873a) by ignoring -Wenum-constexpr-conversion. 

Unfortunately, I'm not savvy enough with this stuff to modify the Makefile to do the ignore and test it. Happy to help any way I can though.

reillylm, 2024-10-26:
Using the link from @jeffdech, I tried manually downloading GDB 10.1 (the version we currently have in our `avr-gdb` formula) and applying the two patches outlined in the issue (manually, not with `git apply`, since the patches had been generated from newer versions of the files than the ones we have in 10.1).

Building and running seemed to work fine; I was able to step thru a simple program built by `avr-gcc`, with GDB in `target remote` mode pointing to simavr.

So hypothetically we could add these patches to our formula. However, I also tried the latest GDB 15.2, and that worked for me too. So I wonder if we should just upgrade our formula to the newer version and match the [upstream gdb formula on homebrew-core](https://github.com/Homebrew/homebrew-core/blob/9a1bd13e6d7e6b4379465900ffbf6afb45d9d205/Formula/g/gdb.rb)? Or do like we have for `avr-gcc` and have separate version-locked formulae? I can volunteer to open a PR either way.

ladislas, 2024-10-27:
If it works I think it makes sense to use the latest version. Sticking to what brew/core is doing is usually the simplest way to go

I'll review your or as soon as you make it 🙂

ladislas, 2024-11-25:
this should be fixed by 30dace29e690270a89fd1bd1ad8983c2464b1b6b
i'll close the issue, feel free to reopen if needed